grblHAL/STM32F1xx #2: 64 pin STM32F103RC with 256 Kbytes of Flash memory?

MeKeCNC, 2021-07-26:
Hi
Is it possible to support, 64 pin STM32F103RC with 256 Kbytes of Flash memory?

terjeio, 2021-07-26:
It should work out of the box? You will have to create a map file for it as I assume none of the current mappings will work.
Additional features above what is currently supported might be a bit problematic as both free RAM and FLASH is on the low end with the current linker file definitions. This can be overcome by adding a new build option and a corresponding linker file for this MCU.

MeKeCNC, 2021-07-26:
Hi terjeio
Thank you very much for your quick reply, I appreciate your work very much. I prefer this card because of its affordable price and many features. I would be very grateful if you could help to create a separate project build file/folder for this processor, I can create the CPU pin map file myself

terjeio, 2021-07-26:
There is no need for a new project build folder, post the pin map file here and I will incorporate it in the current one. Alternatively create a PR for it.

MeKeCNC, 2021-07-26:
I have create the CPU pin map, the file is here as attachment,
[BTT_SKR_MINI_E3_V2.0_map.zip](https://github.com/grblHAL/STM32F1xx/files/6881383/BTT_SKR_MINI_E3_V2.0_map.zip)

terjeio, 2021-07-27:
UART mode for TMC2209 is not possible to test for me as I do not have hardware for that. With lots of luck in might work...

SDCAR and EEPROM support should be possible to add as well, but needs code changes as pins & ports are not the same as the ones already supported?

How is this board programmed, via a bootloader or a ST-link programmer? If via a bootloader it is likely changes will have to be made (new linker script).

MeKeCNC, 2021-07-27:
via bootloader; you put the bin file inside the SD card and when you reboot the board, the code loads

terjeio, 2021-07-27:
> Can we upload the GRBL based compiled bin file/code to the card with the same method?

Yes, but kind of hard for me to make a working binary when I am not able to test it. And there is a risk of bricking the card, the only way to recover it if this happens is to use a ST-Link programmer. Do you have one?

MeKeCNC, 2021-07-27:
Yes I have ST-Link programmer, no problem I can repair the bootloader, by the way I compiled the code you edited and uploaded it to the board, I can communicate and move the motors but I couldn't communicate  and set the tmc2209

terjeio, 2021-07-28:
Great, having the programmer means you can set breakpoints for debugging. Do you have access to an oscilloscope or logic analyzer?

First step would be to check if there is any traffic on the Trinamic driver UART, set breakpoints here
https://github.com/grblHAL/STM32F1xx/blob/643400dcb703fa0c31886aa76a1a99775777e187/Src/serial.c#L427
and here
https://github.com/grblHAL/STM32F1xx/blob/643400dcb703fa0c31886aa76a1a99775777e187/Src/serial.c#L440

Are they hit?

MeKeCNC, 2021-07-28:
Yes, I have both the oscilloscope and the logic probe.
During the debug process, it did not pass this point before reaching the debug points you mentioned. https://github.com/grblHAL/core/blob/2068165e62c949bda37e7006aebd1406b18d76d2/nvs_buffer.c#L244

terjeio, 2021-07-28:
> Yes, I have both the oscilloscope and the logic probe.

Great, that will help if we run into problems with the UART code.

I have done some testing today with a Bluepill. There are a couple of bugs - init code for the TMC2209 was not in place and there were some timing issues in the low-level interface code. I'll commit an update tomorrow.

Using flash for settings storage might require some work, the memory map is different and that could be the reason it hangs. I'll come back with how to disable this. It is likely to be easier to later switch to EEPROM storage than attempt to fix flash storage, I prefer EEPROM storage when available so I guess that should be prioritized.

> Other thing, is there any gcode to check the status of tmc2209 via using console?

Yes, a number of Marlin style M-codes are implemented - M122 is for debugging. Info [here](https://github.com/grblHAL/Plugins_motor/).

FYI [ioSender](https://github.com/terjeio/ioSender) has a tab for Trinamic driver tuning that uses the debug information.

Example M122 output:

```
[TRINAMIC]
                      X
Driver          TMC2209
Set current         500
RMS current         489
Peak current        691
Run current       15/31
Hold current       7/31
CS actual          7/31
stealthChop       false
msteps                2
tstep           1048575
pwm
threshold             0
[mm/s]                -
OT prewarn        false
OT prewarn has
been triggered    false
off time              3
blank time            1
hysteresis
-end                 -2
-start                4
Stallguard thrs       0
DRIVER STATUS:
stallguard
sg_result             0
stst                  *
fsactive
olb
ola
s2gb
s2ga
otpw
ot
STATUS REGISTERS:
 X = 0x80:07:00:00
ok
```

Note that you have to enable Trinamic drivers with the $338 setting!

terjeio, 2021-07-29:
Update just committed. I have enabled EEPROM support - this causes flash overflow so I had to add a new linker script to take advantage of the larger flash (and RAM). You have to select the Debug F103RC build option available in the build menu (from the "hammer" icon) to make it compile.

MeKeCNC, 2021-07-29:
You are a responsible and very helpful person, thank you very much.

> Update just committed. I have enabled EEPROM support - this causes flash overflow so I had to add a new linker script to take advantage of the larger flash (and RAM). You have to select the Debug F103RC build option available in the build menu (from the "hammer" icon) to make it compile.

I re-uploaded the code you just edited, but nothing changed, it still hangs in the same place during debug, I took a video and added it below
https://youtu.be/uuYLsN6Yqkc

> FYI [ioSender](https://github.com/terjeio/ioSender) has a tab for Trinamic driver tuning that uses the debug information.

in ioSender, after connet the bord the console output 
and I am trying the M122 code but the command is not accepted
https://youtu.be/KWL4XMe4LuU

terjeio, 2021-07-29:
Have you enabled the BTT board in my_machine.h? This line has to be commented in:

https://github.com/grblHAL/STM32F1xx/blob/ad3d08b4a277e0b0c2a188bddc337b8b10424ec9/Inc/my_machine.h#L26

MeKeCNC, 2021-07-29:
Yes this my file

// NOTE: Only one board may be enabled!
// If none is enabled pin mappings from generic_map.h will be used.
//#define BOARD_CNC3040
//#define BOARD_CNC_BOOSTERPACK
#define BTT_SKR_MINI_E3_V20
//#define BOARD_MY_MACHINE // Add my_machine_map.h before enabling this!

// Configuration
// Uncomment to enable.

#define USB_SERIAL_CDC       1 // Serial communication via native USB. Comment out for UART communication.
//#define SDCARD_ENABLE      1 // Run gcode programs from SD card, requires sdcard plugin.
//#define KEYPAD_ENABLE      1 // I2C keypad for jogging etc., requires keypad plugin.
//#define ODOMETER_ENABLE    1 // Odometer plugin.
//#define TRINAMIC_ENABLE 2130 // Trinamic TMC2130 stepper driver support. NOTE: work in progress.
#define TRINAMIC_ENABLE 2209
//#define TRINAMIC_ENABLE 5160 // Trinamic TMC5160 stepper driver support. NOTE: work in progress.
//#define TRINAMIC_I2C       0 // Trinamic I2C - SPI bridge interface.
//#define TRINAMIC_DEV       1 // Development mode, adds a few M-codes to aid debugging. Do not enable in production code.
#define EEPROM_ENABLE      1 // I2C EEPROM support. Set to 1 for 24LC16(2K), 2 for larger sizes. Requires eeprom plugin.
//#define EEPROM_IS_FRAM     1 // Uncomment when EEPROM is enabled and chip is FRAM, this to remove write delay.

MeKeCNC, 2021-07-29:
when I followed the debugging step by step, I found that it was hangs at this point, I think it hangs because it is waiting for data on the usb, is it correct?
https://github.com/grblHAL/STM32F1xx/blob/ad3d08b4a277e0b0c2a188bddc337b8b10424ec9/Src/usb_serial.c#L80

terjeio, 2021-07-29:
> when I followed the debugging step by step, I found that it was hangs at this point, I think it hangs because it is waiting for data on the usb, is it correct?

When the USB output buffer is full it loops there until a connection is established.

I have updated the ST driver framework to the latest version and I am having problems with flakiness - perhaps due to the MCU beeing a clone or the fact that I have nearly used all available flash/RAM:

[image]

I have found a couple of further bugs, and I am reminded why did not add TMC2209 to the initialization - the code is not 100% ready... If you are still interested there are a few more things to try:

Add
`#define TRINAMIC_MIXED_DRIVERS 0`
at the start of _btt_skr_mini_e3_2.0_map.h_. This will disable the $338 setting, it is not needed since it is not possible to have other drivers than TMC2209. It will also bypass a problem where the setting $338 does not immediately enable the drivers, a restart is required which, for you, I believe resets $338 back to 0.

Change this line
https://github.com/grblHAL/STM32F1xx/blob/ad3d08b4a277e0b0c2a188bddc337b8b10424ec9/Src/serial.c#L448
to
`USART->BRR = UART_BRR_SAMPLING16(HAL_RCC_GetPCLK1Freq(), 115200);`
to get the correct baud rate.

I am not sure this line is needed or not, comment out if UART output is not working:
https://github.com/grblHAL/STM32F1xx/blob/ad3d08b4a277e0b0c2a188bddc337b8b10424ec9/Src/serial.c#L445

Disable settings storage by changing
https://github.com/grblHAL/STM32F1xx/blob/ad3d08b4a277e0b0c2a188bddc337b8b10424ec9/Src/driver.c#L905
to
`#elif xFLASH_ENABLE`
as it seems the I2C EEPROM is not recognized. I have it working with my Bluepill so I am not sure why. Can you check the label on the EEPROM chip and verify that it is an AT24C32?

Set breakpoints at line 39, 61 and 63 here:
https://github.com/grblHAL/STM32F1xx/blob/ad3d08b4a277e0b0c2a188bddc337b8b10424ec9/Src/btt_skr_mini_e3_2.0.c#L39-L63
When continuing from line 39 you should see output on the UART line, if a valid response is forthcoming the next break should be at line 61 - if not it will stop at line 63.

Here is a valid read transaction on my scope, timebase is 200us/div:
[image]

The blue and yellow traces track each other when transmitting, the blue trace shows the response. This since Rx an Tx is wired together via 1K resistor and I am monitoring on both ends of it.

Finally - I can upload the project as it is now as a zip file if of interest.

MeKeCNC, 2021-07-29:
> Finally - I can upload the project as it is now as a zip file if of interest.

Yes please, I want to run this board with grblHal. I am quite interested it

MeKeCNC, 2021-07-29:
What's your suggestion about it not accepting the M122 command?

terjeio, 2021-07-29:
> Yes please, I want to run this board with grblHal. I am quite interested it

Ok, you can download it from [here](http://www.io-engineering.com/downloads).

> What's your suggestion about it not accepting the M122 command?

It is due to no motors accepting/responding to configuration during startup.

https://github.com/terjeio/Trinamic-library/blob/5e48bda6977b282d2543c41c83e27ecde224f3ff/tmc2209.c#L103-L108

MeKeCNC, 2021-07-29:
> the EEPROM chip and verify that it is an AT24C32?

BT24C32A

terjeio, 2021-07-29:
> BT24C32A

https://github.com/grblHAL/STM32F1xx/blob/ad3d08b4a277e0b0c2a188bddc337b8b10424ec9/Inc/btt_skr_mini_e3_2.0_map.h#L37

Change 2 to 3 in this line  - I forgot about the 4K chips having a 32 byte page size. I believe it is the first time I've encountered one, hopefully EEPROM storage will then work (you will have to undo the FLASH_ENABLE hack above to test).

MeKeCNC, 2021-07-29:
> Change 2 to 3 in this line - I forgot about the 4K chips having a 32 byte page size. I believe it is the first time I've encountered one, hopefully EEPROM storage will then work (you will have to undo the FLASH_ENABLE hack above to test).

EEPROM is OK, stored the my setting

terjeio, 2021-07-29:
> still tmc2209 is not working :(

Are the breakpoints mentioned above above getting hit? If so, any activity on the UART line?

MeKeCNC, 2021-07-29:
> Are the breakpoints mentioned above above getting hit? If so, any activity on the UART line?

I will remove the card from its slot and monitor the uart activities with the analyzer as soon as possible.

terjeio, 2021-08-02:
I have assembled a revised [Nucleo-64 CNC breakout board](https://github.com/terjeio/CNC_Breakout_Nucleo64) (for a F446) and managed to get multiple TMC2209 drivers working by increasing the post read data delay to 5000 (from 150):

https://github.com/grblHAL/STM32F1xx/blob/ad3d08b4a277e0b0c2a188bddc337b8b10424ec9/Src/btt_skr_mini_e3_2.0.c#L65

Note that the current code does not support sensorless homing as the Stallguard implementation is different from  TMC2130/5160 version. I am looking into that now.

MeKeCNC, 2021-08-02:
Sorry for delayed reply
I took the card out, figured out UART pins from schematic and connected the RxTx pins with a cable. I compiled and uploaded the project you sent without any changing it, it still hangs during the debugging process, it still doesn't reach the points you mentioned. Also, the RxTx pins are pulled from zero to high level and staying that way.

[image]

MeKeCNC, 2021-08-02:
When I upload the code to the card and run it normally, while restart process the Rx TX activities are as follows

[image]

terjeio, 2021-08-03:
Have you connected motor power? IIRC the drivers do not respond if not.

MeKeCNC, 2021-08-03:
> Have you connected motor power? IIRC the drivers do not respond if not.

You are right, when you set up the motor connections, UART activity occurred, but only X motor gives driver information.

[VER:1.1f.20210715:]
[OPT:VNMSL,35,512,3,0]
[NEWOPT:ENUMS,RT+,TC,TMC=1]
[FIRMWARE:grblHAL]
[NVS STORAGE:*EEPROM]
[DRIVER:STM32F103C8]
[DRIVER VERSION:210727]
[BOARD:BTT SKR MINI E3 V2.0]
[PLUGIN:Trinamic v0.04]
[GC:G0 G54 G17 G21 G90 G94 G49 G98 G50 M5 M9 T0 F0 S0.]
[G54:0.000,0.000,0.000]
[G55:0.000,0.000,0.000]
[G56:0.000,0.000,0.000]
[G57:0.000,0.000,0.000]
[G58:0.000,0.000,0.000]
[G59:0.000,0.000,0.000]
[G59.1:0.000,0.000,0.000]
[G59.2:0.000,0.000,0.000]
[G59.3:0.000,0.000,0.000]
[G28:0.000,0.000,0.000]
[G30:0.000,0.000,0.000]
[G92:0.000,0.000,0.000]
[TLO:0.000,0.000,0.000]
[PRB:0.000,0.000,0.000:0]
[GC:G0 G54 G17 G21 G90 G94 G49 G98 G50 M5 M9 T0 F0 S0.]
M122
[TRINAMIC]
                      X
Driver          TMC2209
Set current        1000
RMS current         994
Peak current       1405
Run current       17/31
Hold current       8/31
CS actual          8/31
stealthChop       false
msteps               32
tstep           1048575
pwm
threshold             0
[mm/s]                -
OT prewarn        false
OT prewarn has
been triggered    false
blank time            1
hysteresis
-end                 -2
-start                4
Stallguard thrs       0
DRIVER STATUS:
stallguard     
sg_result             0
stst                  *
fsactive               
s2gb                   
s2ga                   
STATUS REGISTERS:
 X = 0x80:08:00:C0

MeKeCNC, 2021-08-03:
Ok. It worked on other drivers
I disabled this command
//#define TRINAMIC_MIXED_DRIVERS 0

[image]

MeKeCNC, 2021-08-03:
While testing the movement of the motors I noticed that the z-axis motor only moves in one direction. When I check the cpu map file of the card, I think this is the source of the problem because the direction pin of the z-axis is in a different port. What is your solution suggestion for this?

// Define step direction output pins.
#define DIRECTION_PORT      GPIOB
#define X_DIRECTION_PIN     12 //PB12
#define Y_DIRECTION_PIN     2  //PB2
#define Z_DIRECTION_PIN     3  //PC3 //********
#define X_DIRECTION_BIT     (1<<X_DIRECTION_PIN)
#define Y_DIRECTION_BIT     (1<<Y_DIRECTION_PIN)
#define Z_DIRECTION_BIT     (1<<Z_DIRECTION_PIN)

Likewise, the enable pin of the A axis is in a different port.
#if N_AXIS > 3
#define A_ENABLE_PIN            2  //PD2 //********
#define A_ENABLE_BIT            (1<<A_ENABLE_PIN)
#endif

terjeio, 2021-08-03:
> stealthChop seems not active on all drivers, how do we activate all of them?

Work in progress... TMC2209 is significantly different from TMC2130 and TMC5160, possibly needing its own motor plugin. I have started looking into this. A pity that the DIAG pins are not longer OD complicating limits handling if you want to use sensorless homing.

> ...the direction pin of the z-axis is in a different port... What is your solution suggestion for this?

> Likewise, the enable pin of the A axis is in a different port (PORTD pin 2).

Bitband support has to be added to the driver for these signals, similar to the F4xx driver. The enable signals already use bitbanding but has a common port definition, each pin has to get its own port definition.

MeKeCNC, 2021-08-03:
> Bitband support has to be added to the driver for these signals, similar to the F4xx driver. The enable signals already use bitbanding but has a common port definition, each pin has to get its own port definition.

I think you are talking about this definition.

#define X_DIRECTION_PORT        GPIOB
#define X_DIRECTION_PIN         12 //PB12
#define X_DIRECTION_BIT         (1<<X_DIRECTION_PIN)
#define Y_DIRECTION_PORT        GPIOB
#define Y_DIRECTION_PIN         2 //PB2
#define Y_DIRECTION_BIT         (1<<Y_DIRECTION_PIN)
#define Z_DIRECTION_PORT        GPIOC
#define Z_DIRECTION_PIN         3 //PC3
#define Z_DIRECTION_BIT         (1<<Z_DIRECTION_PIN)
#define DIRECTION_OUTMODE       GPIO_BITBAND

Do you have any work on this? I checked the F4xx code but I guess I can't get over it.

terjeio, 2021-08-03:
> Do you have any work on this?

I have [uploaded](http://www.io-engineering.com/downloads) a new version of the project, for now untested as do not have the board wired up. Note there are changes in driver.c too, not only the map file.

MeKeCNC, 2021-08-03:
> I have [uploaded](http://www.io-engineering.com/downloads) a new version of the project, for now untested as do not have the board wired up. Note there are changes in driver.c too, not only the map file.

great this worked too :) 
thank you so much

MeKeCNC, 2021-08-03:
the next step will be to run the RC spindle so called Brushless spindle. 
like as;
https://github.com/bdring/Grbl_Esp32/blob/main/Grbl_Esp32/src/Spindles/BESCSpindle.cpp
https://github.com/bdring/Grbl_Esp32/blob/main/Grbl_Esp32/src/Spindles/BESCSpindle.h

Can we run a RC spindle using pwm spindle setting?

terjeio, 2021-08-03:
> Can we run a RC spindle using pwm spindle setting?

See the driver [overview](https://github.com/grblHAL/drivers) - "RC Servo/ESC for spindle" row in the tables , and footnote 13.

MeKeCNC, 2021-08-03:
> See the driver [overview](https://github.com/grblHAL/drivers) - "RC Servo/ESC for spindle" row in the tables , and footnote 13.

Adjusting the variables according to footnote 13. I observed the spindle output with an oscilloscope, the M3 S1000 command activates the spindle output but does not produce pwm, M5 turns the output off completely

[image]

here is my config file

[config.txt](https://github.com/grblHAL/STM32F1xx/files/6925440/config.txt)

MeKeCNC, 2021-08-03:
When I use a different port pin reserved for RGB LED on the board, the PWM output is produced, I think the problem with the PWM settings of PORTC is that PC7 does not output.

terjeio, 2021-08-03:
It is not possible to assign an arbitrary pin as spindle PWM output, only PA8 is supported? A deep dive into the datasheet is required to find out if PC7 is possible to use - and if so which timer this pin can be mapped to.

MeKeCNC, 2021-08-03:
> It is not possible to assign an arbitrary pin as spindle PWM output, only PA8 is supported? A deep dive into the datasheet is required to find out if PC7 is possible to use - and if so which timer this pin can be mapped to.

Yes PA8 is working, the reserved for RGB LED on the board is PA8, no problem, so I will use PA8 as a spindle output,
thank you for your great support

terjeio, 2021-08-03:
>  so I will use PA8 as a spindle output

Great, so no need to modify the code. PC7 _can_ be mapped to Timer 3/Channel 2 but will require code changes. Timer 3 is currently used for step pulse generation so that has to be moved to another timer. The Bluepill I have do not have the PC7 output, I am not going to try to change the code without beeing able to test. And backwards compatibility has to be left in place...

MeKeCNC, 2021-08-03:
> Great, so no need to modify the code. PC7 _can_ be mapped to Timer 3/Channel 2 but will require code changes. Timer 3 is currently used for step pulse generation so that has to be moved to another [...]

you are right.

Have you tried SD card? It gives a compiler error when activated. #define SDCARD_ENABLE      1 // Run gcode programs from SD card, requires sdcard plugin.

Compiler errors:
In file included from ../sdcard/sdcard.c:22:
../sdcard/sdcard.h:51:10: fatal error: ff.h: No such file or directory
   51 | #include "ff.h"
      |          ^~~~~~
compilation terminated.
make: *** [sdcard/subdir.mk:22: sdcard/sdcard.o] Error 1
make: *** Waiting for unfinished jobs....
In file included from ../sdcard/ymodem.c:28:
../sdcard/sdcard.h:51:10: fatal error: ff.h: No such file or directory
   51 | #include "ff.h"
      |          ^~~~~~
compilation terminated.
make: *** [sdcard/subdir.mk:22: sdcard/ymodem.o] Error 1
../Src/diskio.c:23:10: fatal error: ff.h: No such file or directory
   23 | #include "ff.h"
      |          ^~~~~~
compilation terminated.
make: *** [Src/subdir.mk:55: Src/diskio.o] Error 1
In file included from ../Src/driver.c:40:
D:/Works/MeKe PCB/Arduino-VS Code/GRBL Driver STM32F103C8/sdcard/sdcard.h:51:10: fatal error: ff.h: No such file or directory
   51 | #include "ff.h"
      |          ^~~~~~
compilation terminated.
make: *** [Src/subdir.mk:55: Src/driver.o] Error 1
"make -j12 all" terminated with exit code 2. Build might be incomplete.

20:34:59 Build Failed. 9 errors, 0 warnings. (took 983ms)

terjeio, 2021-08-03:
Oops, I forgot about that. Add `defined(STM32F103xE) ||` to line 24 in _grbl/platform.h_ so it becomes:
`
#if defined(STM32F103xB) || defined(STM32F103xE) || defined(STM32F401xC) ||  defined(STM32F407xx) || defined(STM32F411xE) || defined(STM32F446xx) || defined(STM32F756xx)`

It compiles for me when I do that.

MeKeCNC, 2021-08-03:
> Oops, I forgot about that. Add `defined(STM32F103xE) ||` to line 24 in _grbl/platform.h_ so it becomes:

I added this but the number of errors decreased only

[image]

MeKeCNC, 2021-08-03:
Ok, it didn't give a compile error after making the following changes

I modified the following files for path
 sdcard/sdcard.h, 
src/driver.c and 
src/diskio.c

I changed both #include "ffi.h" and #include "diskio.h" 
to: 
#include "../FatFs/ff.h"
#include "../FatFs/diskio.h"

terjeio, 2021-08-03:
One more, you have to add `FatFs` to _Includes_ in _C++ General > Paths and Symbols_ as well.

[image]

I was interrupted by a telephone call so I forgot I did that earlier - getting old...

Luckily the SPI port and pins used for the SD-card is the same as the current code so it should work.

MeKeCNC, 2021-08-03:
It gives sd card connection error

[image]

[image]

terjeio, 2021-08-03:
You fooled me with your map file? This text is just before the pin definitions:

```
// The following defines are not used but defined for reference
// Port init and remap is done by HAL_SPI_MspInit() in stm32f1xx_hal_msp.c
```

Have modified [stm32f1xx_hal_msp.c](https://github.com/grblHAL/STM32F1xx/blob/master/Src/stm32f1xx_hal_msp.c) to match the definitions?
I guess _driver.h_ should be included in this file and definition(s) from that used to select a `HAL_SPI_MspInit` version matching the board.

MeKeCNC, 2021-08-03:
> Have modified [stm32f1xx_hal_msp.c](https://github.com/grblHAL/STM32F1xx/blob/master/Src/stm32f1xx_hal_msp.c) to match the definitions?

I made the changes but the sd card error did not change

void HAL_SPI_MspInit(SPI_HandleTypeDef* hspi)
{
  GPIO_InitTypeDef GPIO_InitStruct = {0};
  if(hspi->Instance==SPI1)
  {
  /* USER CODE BEGIN SPI1_MspInit 0 */

  /* USER CODE END SPI1_MspInit 0 */
    /* Peripheral clock enable */
    __HAL_RCC_SPI1_CLK_ENABLE();

#ifdef BTT_SKR_MINI_E3_V20
       // __HAL_RCC_GPIOA_CLK_ENABLE();
        /**SPI1 GPIO Configuration
        PA5     ------> SPI1_SCK
        PA6     ------> SPI1_MISO
        PA7     ------> SPI1_MOSI
        */
    GPIO_InitStruct.Pin = GPIO_PIN_5|GPIO_PIN_6|GPIO_PIN_7;
    GPIO_InitStruct.Mode = GPIO_MODE_AF_PP;
    GPIO_InitStruct.Speed = GPIO_SPEED_FREQ_HIGH;
    GPIO_InitStruct.Pull = GPIO_PULLUP;
    HAL_GPIO_Init(GPIOA, &GPIO_InitStruct);

#else
    //    __HAL_RCC_GPIOB_CLK_ENABLE();
    /**SPI1 GPIO Configuration    
    PB3     ------> SPI1_SCK
    PB4     ------> SPI1_MISO
    PB5     ------> SPI1_MOSI 
    */

    GPIO_InitStruct.Pin = GPIO_PIN_3|GPIO_PIN_5;
    GPIO_InitStruct.Mode = GPIO_MODE_AF_PP;
    GPIO_InitStruct.Speed = GPIO_SPEED_FREQ_HIGH;
    GPIO_InitStruct.Pull = GPIO_PULLUP;
    HAL_GPIO_Init(GPIOB, &GPIO_InitStruct);

    GPIO_InitStruct.Pin = GPIO_PIN_6;
    GPIO_InitStruct.Mode = GPIO_MODE_INPUT;
    //GPIO_InitStruct.Pull = GPIO_NOPULL;
    HAL_GPIO_Init(GPIOB, &GPIO_InitStruct);

#endif
    __HAL_AFIO_REMAP_SPI1_ENABLE();

  /* USER CODE BEGIN SPI1_MspInit 1 */

  /* USER CODE END SPI1_MspInit 1 */
  }

}

terjeio, 2021-08-04:
> __HAL_AFIO_REMAP_SPI1_ENABLE();

... has to be removed. I have [uploaded](http://www.io-engineering.com/downloads) a new version that sets a symbol for backwards compatibility. And I had to reduce the SPI clock frequency to make it work, this is likely due my test setup with a breadboard and flying leads. You should try to increase it again by reducing the clock divider:

https://github.com/grblHAL/STM32F1xx/blob/ad3d08b4a277e0b0c2a188bddc337b8b10424ec9/Src/diskio.c#L203

MeKeCNC, 2021-08-04:
Excellent, thank you so much
SD card also worked, now it's time to try on the CNC machine, with which variable we adjust the holding current of the motors, I guess with the variable below, I made a change for 240mA but the current did not increase

https://github.com/grblHAL/Plugins_motor/blob/4198fd26c19b6e403858df677a50d0325d512736/trinamic.h#L56

terjeio, 2021-08-04:
If changing the default current you have to reset EEPROM settings after flashing with `$RST=*` or `$RST=&`. It is easier to use `$settings`, also available per axis from the _Settings: Grbl_ tab in ioSender:

[image]

MeKeCNC, 2021-08-05:
Great, I tried with the board for the first time, the result is successful, now we are used as a CNC machine with a 3D board, 

https://youtu.be/f9PZOSTOf88

MeKeCNC, 2021-08-06:
terjeio 
As we talked before, firmware is uploaded to this card with SD card using bootloader. What is your suggestion to upload firmware to this card with the same method? Because when we switch between the two firmwares Marlin and grblHAL, it will be easy and convenient to install via SD card.

terjeio, 2021-08-07:
>  What is your suggestion to upload firmware to this card with the same method?

It will be the same (or similar) as for the [LPC176x](https://github.com/grblHAL/LPC176x) driver? A new build option and a corresponding linker file or [linker scripts](https://github.com/grblHAL/LPC176x/tree/master/linkscripts).

noahwilliamsson, 2021-08-10:
Hey nice work getting this up and running.  I'd like to test this out on a two axis machine.  Is there a way to build this project on Linux, without Eclipse? I tried to follow https://github.com/grblHAL/core/wiki/Compiling-GrblHAL for compiling with Arduino (on macOS) but I realized the board wasn't supported by the built-in board manager.

terjeio, 2021-08-10:
> Is there a way to build this project on Linux, without Eclipse?

Possibly, I have not investigated other options such as PlatformIO or use of command line make.

> The following attempt failed because the default target all in the generated Makefile is empty:

The CMakeLists.txt in the grbl folder is only for the core and is intended for builds of drivers that uses cmake, such as the RP2040. `cmake grbl` uses this file only?
I have no idea what it would take to configure this driver for cmake, I am used to the IDE doing the work for me by generating the required makefiles (there is one master makefile and one for each subdirectory).

MeKeCNC, 2021-08-10:
> It will be the same (or similar) as for the [LPC176x](https://github.com/grblHAL/LPC176x) driver? A new build option and a corresponding linker file or [linker scripts](https://github.com/grblHAL/LPC176x/tree/master/linkscripts).

I checked the CPU linker related parts you mentioned, but I couldn't get over it, it was a bit confusing.

[10 more replies]

closed 2023-08-26 by terjeio