grblHAL/STM32F1xx #5: grblHAL support for MKS Robin E3 V1.0 3D printer board

MeKeCNC, 2021-08-11:
I prepared to add board map for MKS Robin E3 V1.0 which has F103RC variant and uses TMC2209 drivers in UART mode. 
MKS Robin compact and suitable for CNC machine, like as E3 V1.0, BTT SKR E3 V2 (#2)

but according to the schematic, UART pins were used differently for each motor, this design did not seem very effective to me. Is there any limit for grblHAL support? https://github.com/makerbase-mks/MKS-Robin-E3-E3D/blob/master/hardware/MKS%20Robin%20E3%20V1.0_002/MKS%20Robin%20E3%20V1.0_002%20SCH.pdf

[image]

https://www.banggood.com/Makerbase-MKS-Robin-E3-Upgraded-32Bit-Control-Board-Integrate-tmc2209-Support-Uart-Mode-Driver-For-CR-10-Ender-3-3D-Printer-p-1666464.html?akmClientCountry=America&&utm_source=google&utm_medium=cpc_ods&utm_campaign=nancy-content-sdsrm-camera-list-content&utm_content=nancy&gclid=Cj0KCQjw6s2IBhCnARIsAP8RfAiCC4yTmDBJEWxNYXTPtpyRSgtJV3fohBrcWTwN-RYfHUp3LKx5EHwaAiHTEALw_wcB&cur_warehouse=CN

terjeio, 2021-08-11:
> Is there any limit for grblHAL support?

RAM / flash / time and patience for coding and testing? If the pins used has hardware UARTs behind them it should be failry easy to add driver code, if UART code has to be implemented in code then a tad more work (I have never written a software UART so I don't know how much). Pity they did not take advantae of connecting all drivers to the same port as the UART protocol can address up to four drivers.

MeKeCNC, 2021-08-18:
I'm starting to think it's not worth the time for this board, BTT SKR E3 V2 is better option for 3D printer and CNC

closed 2021-09-13 by MeKeCNC

MeKeCNC, 2022-07-22:
> RAM / flash / time and patience for coding and testing? If the pins used has hardware UARTs behind them it should be failry easy to add driver code, if UART code has to be implemented in code then a [...]

Hello, 
@terjeio Is it possible to develop a soft UART for this card?

terjeio, 2022-07-22:
> Is it possible to develop a soft UART for this card?

I think so. Porting the [STM32F4xx](https://github.com/grblHAL/STM32F4xx/blob/master/Src/tmc_uart.c) code should not be too hard.