grblHAL/STM32F1xx #9: GrblHAL support for BTT SKR MINI E3 V1.2 3D printer board

AndWalu, 2021-12-09:
SKR MINI E3 V1.2 3D - difference from v 2.0
- no EEPROM
- some pins relocated 

USB_CONN pin have to be pulled down after usbInit() to enable USB.

[Schematic](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/master/hardware/BTT%20SKR%20MINI%20E3%20V1.2/BTT%20SKR%20mini%20E3%20V1.2sch.pdf)
MAP file:  [btt_skr_mini_e3_1.2_map.zip](https://github.com/grblHAL/STM32F1xx/files/7684970/btt_skr_mini_e3_1.2_map.zip)

"USB_SERIAL_CDC 1", " SDCARD_ENABLE 2" and "TRINAMIC_ENABLE 2209"  works.

I have an oscilloscope, logic analyzer and Black Magic debugger so I can run some tests if needed.

terjeio, 2021-12-09:
There is another significant difference, the Trinamic drivers is controlled by a separate (soft) UART per driver, v2.0 is via a single hardware UART. There are code for soft UART controlled 2209s in the [STM32F4xx repo](https://github.com/grblHAL/STM32F4xx/tree/master/Src) for BTT SKR 1.1 and 2.0 that could be used as a starting point for this driver.

AndWalu, 2021-12-09:
> There is another significant difference, the Trinamic drivers is controlled by a separate (soft) UART per driver, v2.0 is via a single hardware UART. There are code for soft UART controlled 2209s in [...]

Unfortunately, you're right. 
I'll have to go back to my code, I really hate programming.
It means that the 2209 are running on the power up settings now.
Maybe that's enough for me...

Please, close all my issues associated with Trinamic driver.

AndWalu, 2021-12-16:
In my opinion, software UART is not a good idea if we have hardware USART avaliable. 
In my solution I use a GPIO mirroring.
Both writing and reading are done using hardware USART.
The only changes in Trinamic code (tmc2209.c) are resetting the TMC2209 address in the datagram to send because all TMC2209 on this board have address set to 0. 
All code responsible of GPIO mirroring are organized in EXTI interrupt handler.
In the core module (btt_skr_mini_e3_2.0.c, driver.c, serial.c) initialization of GPIO, IRQ handle, and switching of the motor UARTs was added.
I will post the code after completing the tests. 

GPIO mirroring idea:
[image]

DanijelMil, 2022-05-06:
Dear AndWalu, did you have any progress with testing.  I am willing to test you code.

AndWalu, 2022-05-06:
W dniu 07.05.2022 o 00:37, DanijelMil pisze:

> Dear AndWalu, did you have any progress with testing. I am willing to 
> test you code.

Unfortunately, I had to stop all my hobby works...

closed 2023-01-29 by terjeio