grblHAL/STM32F1xx #10: TMC2209 and motor supply voltage loss.

AndWalu, 2021-12-16:
If the the motor supply voltage (VS) of TMC2209 is temporarily switched off and re-established then TMC2209 loads power up default settings even VCC_IO was present. 
If a Reset bit is set in GSTAT register of TMC2209 the TMC2209 would have to be reinitialized.
Maybe adding periodic check of GSTAT  is good idea or ioSender can start this check before sending GCODE?

```
M122
[TRINAMIC]
                      X       Y       Z       A
Driver          TMC2209 TMC2209 TMC2209 TMC2209
...
msteps                8       8       8       8
...
```
```
$$
...
$150=8
$151=128
$152=32
$153=16
...
```

terjeio, 2021-12-16:
Use `M122I` to reinitialize the drivers.

> Maybe adding periodic check of GSTAT is good idea or ioSender can start this check before sending GCODE?

A better idea is to write a plugin and use an input pin for monitoring the motor supply? If going down and later up again issue `M122I` automatically.

AndWalu, 2021-12-16:
Thanks for reminding me of the  ```M122I``` :)
Unfortunately, my BTT SKR Mini E3 board don't has  power good signal. 
But one TL431 is not a problem.
I often disconnect the motor power to fix something on machine.

terjeio, 2021-12-16:
> But one TL431 is not a problem.

It will be cheaper/simpler with a 3.3V zener and one or two resistors?

AndWalu, 2021-12-16:
TL431 is cheap as Zener diode but is very precise.
You can detect voltage drop not only "present or not".
Anyway, the hardware solution depends on what you have in the drawer.

terjeio, 2021-12-20:
FYI I added a [motor power monitor plugin](https://github.com/grblHAL/Templates/tree/master/my_plugin/Motor%20power%20monitor) while debugging ioports support for the ESP32 driver.

closed 2023-01-29 by terjeio