grblHAL/STM32F1xx #11: Compiling for SKR Mini v1.1, failed to use the existing limit switch port assignment

n0th1n6, 2022-01-17:
Hi, I have created a mapping for this board and also updated the build env for platform IO. However, I am not able to reuse the available header for the limit switch as control pins. This board has both min and max limit pins so I try to use the min limit pins for the control. 

limits

`#define X_LIMIT_PORT            GPIOA
#define X_LIMIT_PIN             2 //PA2
#define Y_LIMIT_PORT            GPIOA
#define Y_LIMIT_PIN             1 //PA1
#define Z_LIMIT_PORT            GPIOC
#define Z_LIMIT_PIN             3 //PC3
#define LIMIT_INMODE            GPIO_BITBAND`

and for control

`#define CONTROL_PORT            GPIOC
#define RESET_PIN               2 //PC2
#define FEED_HOLD_PIN           1 //PC1
#define CYCLE_START_PIN         0 //PC0`

So it looks like it is re-using pins even if these pins are in different port. I always get this error from driver.c

`#if DRIVER_IRQMASK != (LIMIT_MASK+CONTROL_MASK+I2C_STROBE_BIT)
#error Interrupt enabled input pins must have unique pin numbers!
#endif`

How can I proceed with this.

terjeio, 2022-01-17:
> #error Interrupt enabled input pins must have unique pin numbers!

This is a limitation that is unique to the STM32 MCUs - it does not matter that the pins are on different ports. Use EXP1 or EXP2 header pins for control inputs instead?

n0th1n6, 2022-01-17:
Thanks, did it and compile ok now. I try to avoid the exp1 and exp2 because I though it might be used for lcd. Anyway, just have to try the actual hardware now. I will close this now. In case someone will need the mapping, I can share it later. I also did some change in driver.c in an attempt to use theexisting headers.

closed 2022-01-17 by n0th1n6