grblHAL/STM32F1xx #12: SKR Mini v1.1: USB Device descriptor request failed

n0th1n6, 2022-01-20:
Hi. I have successfully compiled this driver for the SKR Mini v1.1 after playing around with the mapping. In my_machine.h, USB_SERIAL_CDC is enabled, also in the platformio.ini as a build flags. I can confirm the the firmware is uploaded because the firmware.bin is now renamed to BIGTREETECH.bigtreetech, which is what happens when I upload Marlin firmware. What could possibly causes this error that I am getting here.

[image]

I am not that good in coding but if given some hint then I can try to find the issue.

terjeio, 2022-01-20:
I have fried my only F103 so I cannot check this.

Can you connect with Marlin firmware?
Is it a Microsoft driver that is complaining?

n0th1n6, 2022-01-21:
I can still upload the Marlin firmware and it will work with CNCJs, at least I believe that the board is still ok. I am guessing that either the firmware is not running properly or the native USB driver is not working properly. With Marlin, it uses the maple device driver

Here is my my_machine.h
```c
//#define BTT_SKR_MINI_E3_V20
#define BTT_SKR_MINI_V11
//#define BOARD_MY_MACHINE // Add my_machine_map.h before enabling this!

// Configuration
// Uncomment to enable.

#define USB_SERIAL_CDC       1 // Serial communication via native USB. Comment out for UART communication.
//#define SDCARD_ENABLE      1 // Run gcode programs from SD card, requires sdcard plugin.
```

Here is my platformio build env.
```ini
[env:BTT_SKR_MINI_V11]
board = genericSTM32F103RC
board_build.ldscript = STM32F103RCTX_FLASH.ld
build_flags = ${common.build_flags}
  # See Inc/my_machine.h for options
  -D BTT_SKR_MINI_V11=
  #-D EEPROM_ENABLE=3
  -D USB_SERIAL_CDC=1
  # Relocate the vector table where the boot loader expects to find them
  -D VECT_TAB_OFFSET=0x7000
  -Wl,--defsym=LD_VECT_TAB_OFFSET=0x7000
lib_deps = ${common.lib_deps}
  eeprom
  trinamic
  #USBComposite for STM32F1@==0.91
lib_extra_dirs = ${common.lib_extra_dirs}
# Upload is not supported for this board since BOOT0 is tied to GND.
# With the default boot loader, you must deploy new firmware by copying
# .pio/build/<env name>/firmware.bin (produced by `pio run`) to the SD card.

```

terjeio, 2022-01-21:
You need to change the code a bit and use a different loader script.

Change these lines:
https://github.com/grblHAL/STM32F1xx/blob/942afe00f3c7e385b765d541ffa4571d638ab013/Src/driver.c#L1021-L1026
to:
```
// Initialize HAL pointers, setup serial comms and enable EEPROM
// NOTE: Grbl is not yet configured (from EEPROM data), driver_setup() will be called when done

bool driver_init (void)
{
#ifdef HAS_BOOTLOADER_
    extern uint8_t _FLASH_VectorTable;
    __disable_irq();
    SCB->VTOR = (uint32_t)&_FLASH_VectorTable;
    __DSB();
    __enable_irq();
#endif

    // Enable EEPROM and serial port here for Grbl to be able to configure itself and report any errors
```
and add
`-D HAS_BOOTLOADER`
to your platformio build env.

Add this loader script in the same folder as the original and change platformio build env to use it:

[STM32F103RCTX_BL28K_FLASH .zip](https://github.com/grblHAL/STM32F1xx/files/7910930/STM32F103RCTX_BL28K_FLASH.zip)

With luck that should make it work.

n0th1n6, 2022-01-21:
Thanks @terjeio, got it working now. At least I know that the firmware is uploaded and that the PC and controller are communicating. I will proceed with the actual hardware test later.

```
client> $$
[MSG:'$H'|'$X' to unlock]
$0=10.0 (Step pulse time, microseconds)
$1=25 (Step idle delay, milliseconds)
$2=0 (Step pulse invert, mask)
$3=0 (Step direction invert, mask)
$4=7 (Invert step enable pin, boolean)
$5=0 (Invert limit pins, boolean)
$6=0 (Invert probe pin, boolean)
$7=0
$10=511 (Status report options, mask)
$11=0.010 (Junction deviation, millimeters)
$12=0.002 (Arc tolerance, millimeters)
$13=0 (Report in inches, boolean)
$14=0
$15=0
$16=0
$17=0
$18=0
$19=0
$20=0 (Soft limits enable, boolean)
$21=0 (Hard limits enable, boolean)
$22=0 (Homing cycle enable, boolean)
$23=0 (Homing direction invert, mask)
$24=25.0 (Homing locate feed rate, mm/min)
$25=500.0 (Homing search seek rate, mm/min)
$26=250 (Homing switch debounce delay, milliseconds)
$27=1.000 (Homing switch pull-off distance, millimeters)
$28=0.100
$29=0.0
$30=1000.000 (Maximum spindle speed, RPM)
$31=0.000 (Minimum spindle speed, RPM)
$32=0 (Laser-mode enable, boolean)
$33=5000.0
$34=0.0
$35=0.0
$36=100.0
$37=0
$39=1
$40=0
$43=1
$44=4
$45=3
$46=0
$62=0
$63=2
$64=0
$65=0
$100=250.000 (X-axis travel resolution, step/mm)
$101=250.000 (Y-axis travel resolution, step/mm)
$102=250.000 (Z-axis travel resolution, step/mm)
$110=500.000 (X-axis maximum rate, mm/min)
$111=500.000 (Y-axis maximum rate, mm/min)
$112=500.000 (Z-axis maximum rate, mm/min)
$120=10.000 (X-axis acceleration, mm/sec^2)
$121=10.000 (Y-axis acceleration, mm/sec^2)
$122=10.000 (Z-axis acceleration, mm/sec^2)
$130=200.000 (X-axis maximum travel, millimeters)
$131=200.000 (Y-axis maximum travel, millimeters)
$132=200.000 (Z-axis maximum travel, millimeters)
$140=500
$141=500
$142=500
$150=16
$151=16
$152=16
$200=22.0
$201=22.0
$202=22.0
$210=50
$211=50
$212=50
$220=22.0
$221=22.0
$222=22.0
$339=0
$341=0
$342=30.0
$343=25.0
$344=200.0
$345=100.0
$384=0
ok
GrblHAL 1.1f ['$' or '$HELP' for help]
[MSG:'$H'|'$X' to unlock]
>
```

closed 2023-08-26 by terjeio