grblHAL/STM32F1xx #14: Some step pulses have much longer duration than configured in $0 (STM32F103-based board)

dmitriybabintsev, 2022-02-16:
Hi guys.
First of all I would like to say many thanks to Terje Io and all other guys participated in GRBL development.  This is  a really good job.

Recently I have faced a strange issue with incorrect operation of step motors connected to my STM32F103-based board (STM32F103CBT6). After a couple of hours since I launched my G-code program, I noticed that a distance travelled by a motor does not correspond a distance specified in the executed G-code. Here's my program:
[_all_3.txt](https://github.com/grblHAL/STM32F1xx/files/8078687/_all_3.txt)
The programs consists of multiple repeated movings in Z-axis direction.

Below are my GRBL settings. Step pulse duration and Z-axis acceleration make sense in this test.
[gmsc3u_grbl_backup_2022-02-16_grblHAL.txt](https://github.com/grblHAL/STM32F1xx/files/8078694/gmsc3u_grbl_backup_2022-02-16_grblHAL.txt)

When I started to dig into the issue I noticed that some steps on Z axis have a longer duration than specified in GRBL settings.
Below are some waveforms with different timestamp resolution captured on Z-step pin of my board.

**With 20us resolution:**
[image]

**With 8us resolution:**
[image]

**With 4us resolution:**
[image]

As you can see in the worst case the time interval between 2 steps is less than 2us and looks like some motor drivers detect these 2 pulses as a single step pulse which leads to skipping of steps and therefore errors in travel distance by a motor.

To reproduce the issue it's not necessary to run a program. Just execute a simple command:
G0Z500 
and you will see that lots of pulses have a longer duration than specified in $0 parameter:
[image]

Can you please help to fix this issue.

terjeio, 2022-02-16:
USB or UART connection?
Was the last test run from a sender requesting real-time reports or from a simple terminal program like PuTTY?
In not from a simple terminal program can you rerun the test from one? Issue still visible? If not it is likely interrupt priority for either UART or USB comms has to be lowered.

dmitriybabintsev, 2022-02-16:
I'm using G code sender with USB connection.

I either think that the root cause is in misconfig in interrupt priorities.. Looks like TIM3 interrupt delayed from time to time.

Ok, I will try putty

dmitriybabintsev, 2022-02-16:
It looks much better when connecting with serial terminal.
Below is the waveform after running **G0Z500** command. I still have longer pulses but the difference is much less than previously.

[image]

terjeio, 2022-02-16:
Try reducing the priority for the serial and USB interrupts:

[image]

Increasing the second parameter decreases priority, e.g:

`HAL_NVIC_SetPriority(xxx_IRQn, 1, 0);`

will let the step pulse off interrupt preempt these.

Priority for the systick timer should be reduced as well? IIRC I have done that for a few drivers.

I have fried my only F103 board so I cannot check myself...

dmitriybabintsev, 2022-02-16:
Thanks Terje. I will test it and then update you.

dmitriybabintsev, 2022-02-17:
Hi Terje.
So I lowered USB interrupt priority (USART1 and USART3 are turned off for me) and it's much better now.
```diff
diff --git a/USB_DEVICE/Target/usbd_conf.c b/USB_DEVICE/Target/usbd_conf.c
index f9587a6..1488723 100644
--- a/USB_DEVICE/Target/usbd_conf.c
+++ b/USB_DEVICE/Target/usbd_conf.c
@@ -78,7 +78,7 @@ void HAL_PCD_MspInit(PCD_HandleTypeDef* pcdHandle)
     __HAL_RCC_USB_CLK_ENABLE();
 
     /* Peripheral interrupt init */
-    HAL_NVIC_SetPriority(USB_LP_CAN1_RX0_IRQn, 0, 0);
+    HAL_NVIC_SetPriority(USB_LP_CAN1_RX0_IRQn, 1, 0);
     HAL_NVIC_EnableIRQ(USB_LP_CAN1_RX0_IRQn);
   /* USER CODE BEGIN USB_MspInit 1 */
```
But sometimes when I run the G code program attached previously, I see ~16us pulses (instead of 10us).

[image]

terjeio, 2022-02-17:
Try reducing priority to 2, It should not matter but who knows...
I do not know how long it takes to preempt an USB interrupt, and neither if the USB driver code disables interrupts for critical sections of code. If minimal jitter is a must using a USB <> UART breakout is IMO likely a better choice. Or switch to a faster processor, the F1xx series is pretty slow and does not have a FPU.

dmitriybabintsev, 2022-02-18:
By the way I tried the previous grblHAL version before you moved to a new home:
https://github.com/terjeio/grblHAL/tree/master/drivers/STM32F1xx

and all the step pulses are good there (with USB interrupt priority 1).
So probably there's some "heavy" code on a new version.

Anyway thanks for your help Terje. The issue fixed.

closed 2022-02-18 by dmitriybabintsev