grblHAL/STM32F1xx #15: Support for variety of Mach3 boards based on STM32F1XX

beastieboy, 2022-03-07:
There are many STM32F1 based boards made for Mach3 that can be reprogrammed with grblHAL firmware. Currently I have this one:

[DDMACH V5.0](https://sc04.alicdn.com/kf/Ua101434f1d1945afaafc246f897f9ed9J.jpg)

It's based on STM32F105VCT6. Where to begin with? I can easily trace pinout.

terjeio, 2022-03-07:
> Where to begin with? I can easily trace pinout.

Creating a pin map file would be a good starting point. Then sort out the clock tree configuration. I usually create a processor specific project to verify the USB connection and then copy the clock tree setup from that to the grblHAL driver - I have done so many times for the [STM32F4xx driver.](https://github.com/grblHAL/STM32F4xx/blob/1c8cfaccd3cb45752a5233028086adab79f55178/Src/main.c#L50-L228)

beastieboy, 2022-03-08:
Here are the photos taken together:

[image]
[image]
[image]

terjeio, 2022-09-27:
Did you manage to create a map file?

closed 2023-08-26 by terjeio