grblHAL/STM32F1xx #16: ST USB Middleware makes grblHAL non-free software

StefanBruens, 2022-05-03:
The ST USB code has a really crappy license, which makes it incompatible with GPLed software.

```
  * This software component is licensed by ST under Ultimate Liberty license
  * SLA0044, the "License"; You may not use this file except in compliance with
  * the License. You may obtain a copy of the License at:
  *                      www.st.com/SLA0044
 ```

terjeio, 2022-05-09:
I do not read the licence in that way, but then I am no lawyer...

StefanBruens, 2022-05-09:
Both 4. and 5. are violated/in conflict:
> 4.	This software or any part thereof, including modifications and/or derivative works of this software, must be used and execute solely and exclusively on or in combination with a microcontroller or microprocessor device manufactured by or for STMicroelectronics.
> 5.	No use, reproduction or redistribution of this software partially or totally may be done in any manner that would subject this software to any Open Source Terms. “Open Source Terms” shall mean any open source license which requires as part of distribution of software that the source code of such software is distributed therewith or otherwise made available, or open source license that substantially complies with the Open Source definition specified at www.opensource.org and any other comparable open source license such as for example GNU General Public License (GPL), Eclipse Public License (EPL), Apache Software License, BSD license or MIT license.

terjeio, 2022-05-24:
I'll add the license info from the USB files to the COPYING file. IMO that should be enough.

4. is in compliance as the code cannot be used on other processers unless it is a clone or the code is copied to another project for a different processor.

5. I have not changed the license terms so I cannot see how this comes into play.

StefanBruens, 2022-05-24:
There are plenty of STM32 clones. Unless you can **guarantee** it does not run on any clone (you can't), you are violating the terms.

As soon as the FW is compiled and linked, the complete FW falls under GPL 3.0. Anyone receiving a compiled FW is entitled to receive the complete source code, including the proprietary STM parts. This requirement puts these parts under GPL license conditions.

I think (5.) is absolutely clear STM does not want this, they have made sure there is not a single case left where it can be used in open source software.

The only correct way is to use a USB stack like CherryUSB or TinyUSB.

terjeio, 2022-05-24:
> There are plenty of STM32 clones. Unless you can guarantee it does not run on any clone (you can't), you are violating the terms.

I am not putting the code on a non ST device, the end user may do so. So **I** am not violating the license by making it available.

> I think (5.) is absolutely clear STM does not want this, they have made sure there is not a single case left where it can be used in open source software.

I do not read it like that. I read it as ST does not want the code ported to non ST devices.