grblHAL/STM32F1xx #17: BTT SKR MINI E3 V2- Compiling error when enable SD-CARD

MeKeCNC, 2022-09-19:
The following error occurs, I try to compile when the SD card is activated, including the latest plugins modules for this card.

.........
Compiling .pio\build\BTT_SKR_MINI_E3_V20_USB\src\syscalls.o
Compiling .pio\build\BTT_SKR_MINI_E3_V20_USB\src\sysmem.o
Src\diskio.c:460:9: error: conflicting types for 'disk_read'
 DRESULT disk_read (
         ^~~~~~~~~
In file included from Src\diskio.c:24:0:
FatFS/diskio.h:31:9: note: previous declaration of 'disk_read' was here
 DRESULT disk_read (BYTE pdrv, BYTE* buff, LBA_t sector, UINT count);
         ^~~~~~~~~
Compiling .pio\build\BTT_SKR_MINI_E3_V20_USB\src\system_stm32f1xx.o
*** [.pio\build\BTT_SKR_MINI_E3_V20_USB\src\diskio.o] Error 1
Src\driver.c: In function 'HAL_IncTick':
Src\driver.c:1457:9: warning: implicit declaration of function 'disk_timerproc' [-Wimplicit-function-declaration]
         disk_timerproc();
         ^~~~~~~~~~~~~~
Src\system_stm32f1xx.c:113:0: warning: "VECT_TAB_OFFSET" redefined
 #define VECT_TAB_OFFSET  0x00000000U /*!< Vector Table base offset field.

<command-line>:0:0: note: this is the location of the previous definition
===================================================================================== [FAILED] Took 5.45 seconds =====================================================================================

Environment              Status    Duration
-----------------------  --------  ------------
BTT_SKR_MINI_E3_V20_USB  FAILED    00:00:05.448
=============================================================================== 1 failed, 0 succeeded in 00:00:05.448 =============================================================================== 

 *  The terminal process "C:\Users\myPC\.platformio\penv\Scripts\platformio.exe 'run', '--environment', 'BTT_SKR_MINI_E3_V20_USB'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it.

terjeio, 2022-09-19:
You have to add a search path for include files to the FatFs directory in the .ini file.
And currently the new code will not build unless FatFs is configured for read/write (`#define FF_FS_READONLY 0`) in _ffconf.h_ - I'll fix that in the next release.

MeKeCNC, 2022-09-19:
> You have to add a search path for include files to the FatFs directory in the .ini file. And currently the new code will not build unless FatFs is configured for read/write (`#define FF_FS_READONLY 0`) in _ffconf.h_ - I'll fix that in the next release.

Yes I already have added a search path for include files to the FatFs directory in the .ini file.
also changed (`#define FF_FS_READONLY 0`) in ffconf.h , but still gave compiling error
[platformio.txt](https://github.com/grblHAL/STM32F1xx/files/9600821/platformio.txt)

terjeio, 2022-09-20:
I have commited a fix for the FatFs wrapper and I can build successfully with your platformio.ini above. There are a bunch of warnings but no errors:

```
...
Linking .pio\build\BTT_SKR_MINI_E3_V20_USB\firmware.elf
Checking size .pio\build\BTT_SKR_MINI_E3_V20_USB\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [===       ]  34.4% (used 16896 bytes from 49152 bytes)
Flash: [======    ]  56.3% (used 147660 bytes from 262144 bytes)
Building .pio\build\BTT_SKR_MINI_E3_V20_USB\firmware.bin
```

MeKeCNC, 2022-09-20:
The cause of the problem is due to the FatFs version.
I was trying the latest version of FatFs (FatFs R0.14b), still error, it  worked in previous version (FatFs R0.13c).
Thank you very much for your interest

terjeio, 2022-09-22:
Last commit has been updated for 0.14.

closed 2022-09-22 by terjeio

MeKeCNC, 2022-09-22:
cool, it is working right now