grblHAL/STM32F1xx #22: Not getting any output from USART with or without USB_ENABLE set

dlarue, 2022-09-28:
I figured out what the ioc file is used for and loading it in CubeIDE loaded a visualization of the pin assignments and I see USART_TX on PA9 and USART_RX on PA10.  I built and uploaded the firmware and connected a 3.3v FTDI Rx pin to PA9 and see no grbl output.
I have set compatibility to 1 in config.h and I tried USB_ENABLE to both 0 and 1 and neither results in any grbl header echoed out USART_TX pin.

To test, I uploaded a grbl32 firmware and upon powerup saw grbl setting and version dump. So the board and aux port is operational.

CubeIDE Symbols:
       <name>USE_HAL_DRIVER
        <name>STM32F103xB
        <name>RX_BUFFER_SIZE 512
        <name>NVS_SIZE 1536
        <name>NO_SETTINGS_DESCRIPTIONS
        <name>OVERRIDE_MY_MACHINE
        <name>BOARD_CNC3040
        <name>USB_ENABLE 1
        <name>N_AXIS 3

dlarue, 2022-09-28:
I noticed in my_machine.h there was "#define USB_SERIAL_CDC 1" and comment stating to comment it out for UART comm
but that made no difference. I also set USB_ENABLE = 0 in the symbols but will try with that back at 1.

terjeio, 2022-09-28:
The symbol `USB_ENABLE` is no longer used in grblHAL due to conflict with some other frameworks, use `USB_SERIAL_CDC` instead as you have discovered. When `USB_SERIAL_CDC`  is not defined or set to 0 UART comms should be default. One way to check this is to see which part of the code gets compiled in _driver.c_:

[image]

Here you can see `usbInit()` is greyed out  and `serialInit()` is active. Note that sometimes the greyed out parts might be wrong, right click on the project folder in the IDE and select _Index > Rebuild_ to be sure it is correct.

Do you have a ST-Link and can set breakpoints? `serialWriteS()` in _serial.c_ is the first place to check, is it ever called? If not has the board gone into a hardfault? Press the pause button to check if so.

dlarue, 2022-09-28:
I checked and it is flowing through stream_connect(serialInit()) and to be sure I removed USB_ENABLE and set USB_SERIAL_CDC 0 in my_machine.h and still nothing on the serial output.

I am using ST-Link to upload firmware but don't know yet how to use it to set breakpoints. That would be very handy to have working. Currently I'm only using the ST-Link via OpenOCD to upload code. I'll see if I can figure out how to get debugging happening.

terjeio, 2022-09-28:
>  I removed USB_ENABLE and set USB_SERIAL_CDC 0 in my_machine.h

I assume you removed OVERRIDE_MY_MACHINE from the IDE symbols then...

To set a breakpoint just double click to the left of the line number:

[image]

dlarue, 2022-09-28:
I was just covering my ars by making sure it was in both places. :-)

I have debugging working and it's breaking at line 103 in serialWriteS()

dlarue, 2022-09-28:
Seems to be wanting to print "error:7"

terjeio, 2022-09-28:
error 7 is failure to read settings, either from flash or EEPROM - I belive I saw that this board has an EEPROM mounted, but possibly interfaced via SPI? grblHAL supports I2C EEPROMs.

If you send a '?' do you get new output in serialWriteS?
You can also set breakpoints in the USART1_IRQHandler to see if RX and TX interrupts fires.

dlarue, 2022-09-28:
I'm testing this on a Bluepill board so no external EEPROM. I will connect up my FTDI board(3.3V) and minicom and see if there's any reaction.

dlarue, 2022-09-28:
ok so this is different. It's in debug mode and I connected the FTDI board and now I'm getting stuff!

?
error:7

Grbl 1.1f ['$' for help]
[MSG:'$H'|'$X' to unlock]
?<Alarm|MPos:0.000,0.000,0.000|Bf:35,512|FS:0,0|Pn:PXYZRHS|Ov:100,100,100>

dlarue, 2022-09-28:
it will run through the debugger either single stepping or just running but it will not run when powered via USB.

closed 2022-09-28 by dlarue