grblHAL/STM32F1xx #24: PA1 for spindle pwm out ?

MeKeCNC, 2022-10-24:
Can we use PA1 instead of PA8 as spindle pwm output?

terjeio, 2022-10-24:
Seems it is possible, from PA1 spec: `USART2_RTS(9)/ADC12_IN1/TIM2_CH2(9)`
[driver.h](https://github.com/grblHAL/STM32F1xx/blob/54d0e2d98eaaf7cb6f41665049b948d0a4d28175/Inc/driver.h#L103-L123) must be modified for that, `STEPPER_TIMER` has to be changed as well as it is mapped to timer 2.

You can do it?

MeKeCNC, 2022-12-02:
> Seems it is possible, from PA1 spec: `USART2_RTS(9)/ADC12_IN1/TIM2_CH2(9)` [driver.h](https://github.com/grblHAL/STM32F1xx/blob/54d0e2d98eaaf7cb6f41665049b948d0a4d28175/Inc/driver.h#L103-L123) must [...]

I have modified the driver.h file for SPINDLE PWM on PA1, but SPINDLE_PWM_TIMER_INV gives output when set to 1, output does not work when set to 0

// Define timer allocations.
#define STEPPER_TIMER TIM5
#define PULSE_TIMER TIM3
#define DEBOUNCE_TIMER TIM4

#ifdef SPINDLE_PWM_PORT_BASE

#if SPINDLE_PWM_PORT_BASE == GPIOA_BASE
    #if SPINDLE_PWM_PIN == 1 // PA1 - TIM2_CH2
    #define SPINDLE_PWM_TIMER_N     2
    #define SPINDLE_PWM_TIMER_CH    2
    #define SPINDLE_PWM_TIMER_INV   1
    #define SPINDLE_PWM_AF_REMAP    0
  #endif
  #if SPINDLE_PWM_PIN == 8 // PA8 - TIM1_CH1
    #define SPINDLE_PWM_TIMER_N     1
    #define SPINDLE_PWM_TIMER_CH    1
    #define SPINDLE_PWM_TIMER_INV   0
    #define SPINDLE_PWM_AF_REMAP    0
  #endif
#elif SPINDLE_PWM_PORT_BASE == GPIOB_BASE
  #if SPINDLE_PWM_PIN == 0 // PB0 - TIM1_CH2N
    #define SPINDLE_PWM_TIMER_N     1
    #define SPINDLE_PWM_TIMER_CH    2
    #define SPINDLE_PWM_TIMER_INV   1
    #define SPINDLE_PWM_AF_REMAP    0b01
  #endif
#endif

[image]

terjeio, 2022-12-04:
> I have modified the driver.h file for SPINDLE PWM on PA1, gives output when SPINDLE_PWM_TIMER_INV set to 1, output does not work when set to 0

That is a bit odd, is this with a cloned MCU or genuine one?

> I want normal output of this signal

The scope trace does not tell me anything. What is wrong with it? Should it be 5KHz?

MeKeCNC, 2022-12-04:
> The scope trace does not tell me anything. What is wrong with it? Should it be 5KHz?

I guess I didn't fully explain the problem. The output of the oscilloscope indicates that the output of the generated signal is inverted. The expected signal should be invertesd to the signal on the oscilloscope, how should we set it up?

The generated PWM signal should be a signal for the BESC spindle and the normal inverse of what appears on the oscilloscope.

MeKeCNC, 2022-12-04:
> That is a bit odd, is this with a cloned MCU or genuine one?

real mcu, i am using BTT SKR MINI E3 V2.0 3D printer board

terjeio, 2022-12-04:
$16=4 works? ($$=16 for info)

MeKeCNC, 2022-12-04:
> `$16=4` works? (`$$=16` for info)

$$=16
    0 - Spindle enable (1)
    1 - Spindle direction (2)
    2 - PWM (4)
Inverts the spindle on, counterclockwise and PWM signals (active low).

MeKeCNC, 2022-12-04:
> #define SPINDLE_PWM_TIMER_INV 1

when  SPINDLE_PWM_TIMER_INV  set to 0
#define SPINDLE_PWM_TIMER_INV 0

no pwm output on PA1

[image]

terjeio, 2022-12-05:
There is a nasty bug [here](https://github.com/grblHAL/STM32F1xx/blob/db90562953825b60d215abe9f97d4e2d21c4cbc0/Src/driver.c#L592), wonder if it is this that trips it up for PA1? The line should be:

`SPINDLE_PWM_TIMER->CCER |= SPINDLE_PWM_CCER_EN;`

MeKeCNC, 2022-12-06:
> There is a nasty bug [here](https://github.com/grblHAL/STM32F1xx/blob/db90562953825b60d215abe9f97d4e2d21c4cbc0/Src/driver.c#L592), wonder if it is this that trips it up for PA1? The line should be:
> 
> `SPINDLE_PWM_TIMER->CCER |= SPINDLE_PWM_CCER_EN;`

good catch, it works now. that is, we can use two different pins for the pwm spindle output on the BTT SKR MINI E3 V2.0 3D printer board. in driver.h latest revision in driver.h code as below

// Define timer allocations.
#define STEPPER_TIMER TIM2
#define PULSE_TIMER TIM3
#define DEBOUNCE_TIMER TIM4

#ifdef SPINDLE_PWM_PORT_BASE

#if SPINDLE_PWM_PORT_BASE == GPIOA_BASE
    #if SPINDLE_PWM_PIN == 1 // PA1 - TIM5_CH2
    #define SPINDLE_PWM_TIMER_N     5
    #define SPINDLE_PWM_TIMER_CH    2
    #define SPINDLE_PWM_TIMER_INV   0
    #define SPINDLE_PWM_AF_REMAP    0
  #endif
  #if SPINDLE_PWM_PIN == 8 // PA8 - TIM1_CH1
    #define SPINDLE_PWM_TIMER_N     1
    #define SPINDLE_PWM_TIMER_CH    1
    #define SPINDLE_PWM_TIMER_INV   0
    #define SPINDLE_PWM_AF_REMAP    0
  #endif
#elif SPINDLE_PWM_PORT_BASE == GPIOB_BASE
  #if SPINDLE_PWM_PIN == 0 // PB0 - TIM1_CH2N
    #define SPINDLE_PWM_TIMER_N     1
    #define SPINDLE_PWM_TIMER_CH    2
    #define SPINDLE_PWM_TIMER_INV   1
    #define SPINDLE_PWM_AF_REMAP    0b01
  #endif
#endif