grblHAL/STM32F1xx #30: Usb not working on 103RC

r3l4x-pt, 2023-01-14:
Im trying to use grblhal on mach3 usb board. It has a stm32f103rct6 and needs pulldown on PC11.
i managed to work out the PC11 but i always get usb descriptor failed.
I already created a simple program using cubemx to test the usb and it works. I get a com port.

The board is 
[image]

r3l4x-pt, 2023-01-15:
Ive managed to make it work by compiling for the blue pill. Usb works and using UGS left it running about 2 hours and it didn't crash. Is there a drifrence in the code between ghe C8 and the RC?

terjeio, 2023-01-15:
> Is there a diffrence in the code between the C8 and the RC?

No - not in the grblHAL code, and both uses the same clock tree configuration.
An issue with the board map? Which one did you use that failed?
The RC build versions sets a different MCU symbol that may change some of the STM library code so it could be due to that as well.

Did you make a new board map for the board? If so can you share it? IIRC somebody requested one earlier but I do not have a board available for testing.

r3l4x-pt, 2023-01-15:
This is the steps ive done

Import project
Add __HAL_RCC_GPIOD_CLK_ENABLE(); to main.c
Add my_machine.c to Src folder
Add my_machine_map.h to Inc folder
Uncomment BOARD_MY_MACHINE in my_machine.h
Add N_AXIS 4 to symbols
Add a missing }, on line 118 of driver.c
Build Release 

everything is working but If i build Release F103RC i get Unknown USB Device (Device Descriptor Request Failed)

attached is the my_machine_map.h and my_machine.c

[mach3_map.zip](https://github.com/grblHAL/STM32F1xx/files/10421311/mach3_map.zip)

also the chip in this board its marked as STM32F103RCT6 but CubeIDE says its a fake  and i belive its a GD32 chip as many boards ive seen on aliexpress have it

r3l4x-pt, 2023-01-17:
changing the  #define FLASH_BANK1_END in stm32f103xe.h from 0x0807FFFFUL  to  0x0803FFFFUL  makes usb work.

r3l4x-pt, 2023-01-17:
How can i use a DRO with grblhal?
The board has connection only for UART5

terjeio, 2023-01-18:
Thanks for the map, I have added it as BOARD_MACH3_BOB and fixed the other bits as well. The flash size is not something I want to change though, looking for a variant with the correct properties and flash size is the way to go?

> How can i use a DRO with grblhal?
The board has connection only for UART5

By duplicating the real-time report to UART5?
Either serial.c has to be modified for UART5 or UART5 support has to be added to the board code. If you want data in a specific format that has to be added to the board code as well.

r3l4x-pt, 2023-01-18:
This board has a fake chip and most likely doesnt have 256k of flash or its not a 103RCT6.

For the DRO the i wnted to duplicate the report so i could connect uart5 to a another mcu to process the report and display it to a HMI screen. The documentation says the keypad and DRO can be UART OR I2C but how do i set it to UART?

terjeio, 2023-01-18:
> This board has a fake chip and most likely doesnt have 256k of flash or its not a 103RCT6.

Yes, the issue is how to handle that?

> The documentation says the keypad and DRO can be UART OR I2C but how do i set it to UART?

`#define KEYPAD_ENABLE 2`
 https://github.com/grblHAL/STM32F1xx/blob/3783f3b1500e88841bc6c60e264370b9de9610fc/Inc/my_machine.h#L39

This is for keypad input, for output MPG mode has to be added to the driver - code can likely be copied verbatim from the STM32F4xx driver. It is guarded by the MPG_MODE symbol.
FYI my F103 bluepill sadly let out the magic smoke so I do not currently have a board to verify any code changes myself.

r3l4x-pt, 2023-01-18:
> > This board has a fake chip and most likely doesnt have 256k of flash or its not a 103RCT6.
> 
> Yes, the issue is how to handle that?
> 
The best way is to add a comment in my_machine.c saying if usb fails in the mach3 board to change the value in FLASH_BANK1_END to 0x0803FFFFUL.

For the uart ive tried to copy from STM32F4xx but cant make it work. I also tried to modify the code from SERIAL2_MOD in serial.c to use uart5 instead of usart3 but still cant get nothing out os uart5. 

I going to try and solder wires in usart1 and 3 to see if they work.

r3l4x-pt, 2023-01-21:
i give up. cant get anything on any report out of uart1 or 5.

tiago928, 2023-01-27:
Did you managed to get uart working.
I have a bluepill and i need uart and usb

r3l4x-pt, 2023-01-28:
No i tried coping the from the f4 but dindnt work. The code compiled without error nothing came out of uart is USB CDC is enabled

terjeio, 2023-01-28:
> I have a bluepill and i need uart and usb

UART for what purpose? UART 1 and 3 have code support, UART1 only when USB is disabled as the code is now, UART 3 when SERIAL2_MOD is defined in _my_machine.h_. _serial.c_ and _driver.c_ should be updated to register the UARTs with the core for greater flexibility - but I am hesitant to do that myself as I cannot verify code changes due to having no boards to test with. The one I had let out its magic smoke.

Anyone willing to donate a board to me? Preferably a BSMCEO4U-PP Mach4 BOB with a F103RC MCU. I can then do the coding and testing. Be aware that the C8 variant (with 128K flash) is on its last legs for grblHAL support, there is very little memory left for both code and data. When memory capacity is exceeded I will tag the last usable version and leave it at that.

tiago928, 2023-01-28:
I need the usb to connect to pc and uart to connect to another device to read data and send commands. My understanding is that uart 1 or 3 is used instead of usb and i need both

r3l4x-pt, 2023-01-28:
> I need the usb to connect to pc and uart to connect to another device to read data and send commands. My understanding is that uart 1 or 3 is used instead of usb and i need both

the bluepill probaly doenst have enough flash size to add code. just disable usb and use another bluepill programed for  SerialPassthrough and you can have usb and uart.

> Anyone willing to donate a board to me? Preferably a BSMCEO4U-PP Mach4 BOB with a F103RC MCU. I can then do the coding and testing.

i can donate

r3l4x-pt, 2023-01-29:
@terjeio i can donate a mach3 board. email me at motex_fcp@hotmail.com

r3l4x-pt, 2023-02-16:
@terjeio did you recived the board?

terjeio, 2023-02-16:
Yes, it came in the post yesterday.

terjeio, 2023-02-17:
@r3l4x-pt  Bad luck - I managed to kill the MCU when I plugged in my USB <> UART breakout. But since I recently ordered a few genuine F103CBs that arrived yesterday I was able to test some of the code changes I have made with a Bluepill. Why I did not order some F103RCs at the same time I cannot explain, will do so and replace the processor so I will be able to debug F103 code again...

I have added UART5 code in _serial.c_ that compiles, hopefully it works. If it does not then the first thing to change is the [clock source](https://github.com/grblHAL/STM32F1xx/blob/ed3d2e1aecab277a3140f369d3fd52a7b345ae9f/Src/serial.c#L45) to `HAL_RCC_GetPCLK1Freq`.

I have also added support for MPG mode that allows switching between USB and UART inputs. This can also be enabled for Cx builds by adding a pin definition for the [mpg mode input](https://github.com/grblHAL/STM32F1xx/blob/ed3d2e1aecab277a3140f369d3fd52a7b345ae9f/Inc/mach3_bob_map.h#L140-L143) and enabling MPG_MODE in _my_machine.h_ However, no board maps has been modified for that yet.

Note that I have changed [pin mappings for the 10pin IDC header](https://github.com/grblHAL/STM32F1xx/blob/ed3d2e1aecab277a3140f369d3fd52a7b345ae9f/Inc/mach3_bob_map.h#L26-L41) and changed mapping comments to match the board labeling. I have also corrected the [CONTROL_INMODE](https://github.com/grblHAL/STM32F1xx/blob/ed3d2e1aecab277a3140f369d3fd52a7b345ae9f/Inc/mach3_bob_map.h#L134) as this was wrong - if shifting is to be used the pins has to be in the same order as the underlying data structure.

@tiago928 - It is possible to enable two UART ports at the same time for the Cx variants, but perhaps not when USB is enabled without tuning RAM use, either in the [linker file](https://github.com/grblHAL/STM32F1xx/blob/ed3d2e1aecab277a3140f369d3fd52a7b345ae9f/STM32F103C8TX_FLASH.ld#L58-L59) or by reducing [buffer sizes](https://github.com/grblHAL/core/blob/2779c833fdf59a0a39f64110ca6e580cfaecae7f/stream.h#L51-L57). Buffer sizes can be set in the project properties by adding the symbols and desired values.

I hope this helps.

terjeio, 2023-02-17:
> Did you have to change the FLASH_BANK1_END to 0x0803FFFFUL to make USB working?

No, I did not change this and I do not remember now if I tested USB as I wanted to get UART5 working - easier to debug via UART when normal debugging does not work. The first time I flashed the LED lit up though so it the code did start.

I am not 100% sure if the resistors will affect UART operation, but I think it is likely.

r3l4x-pt, 2023-02-17:
had to change the clock source to HAL_RCC_GetPCLK1Freq and UART5 is working fine without removing the resistors. one thing that happened was it was stuck in error 18 but after sending $14=1 its working fine.

terjeio, 2023-02-18:
I have commited a change to EEPROM emulation that hopefully fixes the USB issue with FLASH_BANK1_END, I have also added the board to the [Web Builder](http://svn.io-engineering.com:8080/). 

Thanks for the board and time you have put into this - appreciated.

yoroboticscolombia, 2023-04-18:
Hi, I was able to flash couple boards, however I realised it works with the ones that come with the external clock with 8Mhz, the ones that have a clock of 12Mhz, it says usb descriptor not recognised, I followed some comments in this trhread, but unable to get it working, would you recommend any additional step?

Thanks.

terjeio, 2023-04-19:
>  the ones that have a clock of 12Mhz, it says usb descriptor not recognised, I followed some comments in this trhread, but unable to get it working, would you recommend any additional step?

Try with `RCC_PLL_MUL6`, USB needs a 48MHz clock:

https://github.com/grblHAL/STM32F1xx/blob/5cbbdd0f86745ecef9dc502529a841e666f51125/Src/main.c#L60

yoroboticscolombia, 2023-04-20:
> Try with `RCC_PLL_MUL6`, USB needs a 48MHz clock:
>
> https://github.com/grblHAL/STM32F1xx/blob/5cbbdd0f86745ecef9dc502529a841e666f51125/Src/main.c#L60

Thank you, I will try tomorrow, also in this part, its not needed to put the value of 12MHz?

https://github.com/grblHAL/STM32F1xx/blob/5cbbdd0f86745ecef9dc502529a841e666f51125/Src/system_stm32f1xx.c#L96

#if !defined  (HSE_VALUE) 
  #define HSE_VALUE               8000000U /*!< Default value of the External oscillator in Hz.

Thanks.

terjeio, 2023-04-20:
> Thank you, I will try tomorrow, also in this part, its not needed to put the value of 12

Good question, I am not sure I use any calls in this driver where this is referenced. Adding the value to a new build configuration is the best way, an example:

[image]

yoroboticscolombia, 2023-04-21:
Thank you, it worked like a charm, do you know how to edit the usb to make a custom name when it installs the virtual comm, thanks.

terjeio, 2023-04-27:
> do you know how to edit the usb to make a custom name when it installs the virtual comm, thanks.

The name is defined [here](https://github.com/grblHAL/STM32F1xx/blob/d049e6886ae3283553e854c2831fbb64dbf6c257/USB_DEVICE/App/usbd_desc.c#L70).

terjeio, 2023-05-05:
I'll add a 12 MHz crystal build option in the next commit.

closed 2023-05-05 by terjeio