grblHAL/STM32F1xx #31: Port restrictions for 5-axis?

macbef, 2023-01-26:
Not an issue, more like a question.

I'm building a cheap board with an STM32F103CB. I'd like to have 5 axis.
On my board I'm using GPIOA for axis XYZA (Step and Dir), and GPIOB-0 and GPIOB-1 for the B-axis.
Jogging the B-axis results in a movement of X (which is hooked up to GPIOA0 and GPIOA1).

Is there a restriction in the usage of the ports? Do all STEP-pins have to be on the same port?
(The same with DIR-pins?).

And I tried to map the spindle-PWM to GPIOB-7, but that doesn't compile.

Maybe I should live with the CNC3040 pinout. Seems to compile fine.

terjeio, 2023-01-26:
First, you should not base new designs on the CB - it may soon drop out of support due to limited memories. Go for the RC version.

> Is there a restriction in the usage of the ports? Do all STEP-pins have to be on the same port?

No, if using GPIO_BITBAND as the OUTMODE.  See the new [Mach3 BOB map](https://github.com/grblHAL/STM32F1xx/blob/ec3bf8a36e99a34023ba9ab941cc3e74bfeec8e8/Inc/mach3_bob_map.h#L54-L70) for how to.

> And I tried to map the spindle-PWM to GPIOB-7, but that doesn't compile.

If B7 can be "wired" to a free timer then you may try adding it to [driver.h definitions for PWM](https://github.com/grblHAL/STM32F1xx/blob/ec3bf8a36e99a34023ba9ab941cc3e74bfeec8e8/Inc/driver.h#L113-L137).

macbef, 2023-01-26:
Thanks, I'll look into these examples.
I didn't know that the F103RC's are supported in grblHAL. Haha, but now that I look at the platformio.ini ... The BTT_SKR_MINI_E3_V20 uses an RC. I'll move my design to this sub-family. (RB's are not supported? Same controller with a little less memory)

closed 2023-01-26 by macbef