grblHAL/STM32F1xx #32: STM32CubeIDE Build Project failed with 5 errors.

layerswing, 2023-02-19:
Hi, 
i've tried to build the GRBL Driver STM32F103C8 in **STM32CubeIDE**. The Build Project failed with 5 errors. 
I can build the project with **pio run --environment bluepill_f103c8_128k_USB**
I've attached the whole output.
Best wishes:
Dan

../Src/sysmem.c:63:1: error: unknown type name 'caddr_t'
   63 | caddr_t _sbrk(int incr)
	  | ^~~~~~~
../Src/sysmem.c: In function '_sbrk':
../Src/sysmem.c:76:25: error: 'caddr_t' undeclared (first use in this function)
   76 |                 return (caddr_t) -1;
	  |                         ^~~~~~~
../Src/sysmem.c:76:25: note: each undeclared identifier is reported only once for each function it appears in
../Src/sysmem.c:81:25: error: expected ';' before 'prev_heap_end'
   81 |         return (caddr_t) prev_heap_end;
	  |                         ^~~~~~~~~~~~~~
	  |                         ;
../Src/sysmem.c:67:15: warning: variable 'prev_heap_end' set but not used [-Wunused-but-set-variable]
   67 |         char *prev_heap_end;
	  |               ^~~~~~~~~~~~~
../Src/sysmem.c:82:1: warning: control reaches end of non-void function [-Wreturn-type]
   82 | }
	  | ^
arm-none-eabi-gcc "../Src/usb_serial.c" -mcpu=cortex-m3 -std=gnu11 -g3 -DUSE_HAL_DRIVER -DSTM32F103xB -DDEBUG -DRX_BUFFER_SIZE=512 -DNVS_SIZE=1536 -DNO_SETTINGS_DESCRIPTIONS -c -I../Inc -I../grbl -I../Drivers/CMSIS/Device/ST/STM32F1xx/Include -I../Drivers/CMSIS/Include -I../Drivers/STM32F1xx_HAL_Driver/Inc/Legacy -I../Drivers/STM32F1xx_HAL_Driver/Inc -I../USB_DEVICE/App -I../USB_DEVICE/Target -I../Middlewares/ST/STM32_USB_Device_Library/Core/Inc -I../Middlewares/ST/STM32_USB_Device_Library/Class/CDC/Inc -I../FatFs -I"/home/bkb/workspaces/STM32CubeIDE/GRBL Driver STM32F103C8" -Og -ffunction-sections -fdata-sections -Wall -fstack-usage -MMD -MP -MF"Src/usb_serial.d" -MT"Src/usb_serial.o" --specs=nano.specs -mfloat-abi=soft -mthumb -o "Src/usb_serial.o"
make: *** [Src/subdir.mk:61: Src/sysmem.o] Error 1
make: *** Waiting for unfinished jobs....
"make -j8 all" terminated with exit code 2. Build might be incomplete.

14:30:49 Build Failed. 5 errors, 3 warnings. (took 1s.285ms)

**OS version**
-Version-
Kernel		: Linux 5.19.0-32-generic (x86_64)
Version		: #33~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Mon Jan 30 17:03:34 UTC 2
C Library		: GNU C Library / (Ubuntu GLIBC 2.35-0ubuntu3.1) 2.35
Distribution		: Ubuntu 22.04.2 LTS

**STM32CubeIDE version**
Version: 1.11.2
Build: 14494_20230119_0724 (UTC)
[build_errors.txt](https://github.com/grblHAL/STM32F1xx/files/10777027/build_errors.txt)

terjeio, 2023-02-19:
`caddr_t` is defined in a file in the CubeIDE installation directory:

`/STM32CubeIDE/plugins/com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.10.3-2021.10.win32_1.0.100.202210260954/tools/arm-none-eabi/include/sys/types.h`

Your installation is corrupt or somehow missing a path to this file? Default is to use the workspace settings to locate this file, perhaps this is pointing to a different toolchain?

[image]

FYI I am using 1.11.2 on Windows.

layerswing, 2023-02-19:
Thank you,
This was the solution. I find the **./STMicroelectronics/stm32cubeide_1.11.0/plugins/com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.10.3-2021.10.linux64_1.0.100.202210260954/tools/arm-none-eabi/include/sys/types.h** file, and after changing the settings CubeIde builds the project.
[image]

closed 2023-02-19 by layerswing