grblHAL/STM32F1xx #33: Creality 4.2.2 stm32f103ret6 support

macmimy, 2023-04-01:
I have a bunch of Creality 4.2.2 stm32f103ret6 boards from upgrading ender 3v2's I'm sure I'm not alone, was wondering if its possible to get grblhal on them they seem similar to the blue pill. I found schematics for it https://klipper.discourse.group/t/creality-board-4-2-2-and-4-2-7-schematics/3104 , I'm assuming a pin remap wouldn't be all that is needed ?

terjeio, 2023-04-02:
> I'm assuming a pin remap wouldn't be all that is needed ?

It is all that is needed for basic functionality as the Trinamic drivers are not controlled by code, you can make one?
Note that native USB is not supported - `#define USB_SERIAL_CDC       1` has to be commented out in _my_machine.h_

There is an EEPROM on the board but according to the scematic is not connected to I2C capable pins, if this is to be used code has to be added to bit-bang the I2C protocol.

macmimy, 2023-04-02:
hmm I might give it a shot

nionio6915, 2023-12-17:
I am in the same situation- any update/progress in this?

stevemartinsuk, 2024-01-04:
I have just ported to the Creality 4.2.7 board and successfully run it in coreXY. I have a bit more testing to do, but i'd be happy to upload it.

macmimy, 2024-01-12:
NICE WORK ! I will be giving it a shot, I have 2 boards sitting here

Ferz82, 2024-05-08:
Hi. Has anyone done this with version 4.2.2?

kamil1l, 2025-03-16:
Hi. I have a Creality 4.2.2 with GD32F303 processor and I have a problem. Using web builder, I get the firmware and program it using ST-Link. Everything works fine except for the X and Y axes. The X axis has no DIR signal, and the Y axis doesn’t move completely. The Z-axis (ganged) works perfectly. I want to check if the board is not broken by going back to Marlin, but when I program Marlin via ST-Link, it doesn't start up. Can somebody help?

terjeio, 2025-03-17:
@kamil1l - there was a typo in the map file, hopefully fixed now. A bit odd that F103 code works with a F303, perhaps since yours is a clone?

> Additionally, Marlin compiled for the STM32F103RET6 worked without any issues earlier.

The grblHAL firmware likely overwrote a bootloader.

kamil1l, 2025-03-17:
@terjeio THANK YOU SO MUCH! I struggled with this for 10 hours, and it just wouldn’t work.

I'm not sure, but if you changed #ifdef BOARD_CREALITY_V2_2_7 in the board map file, you also need to update #define BOARD_CREALITY_V2_2_2 to 4_2_2 in my_machine.h. Additionally, in platformio.ini, which is used by the web builder, there is a flag -D BOARD_CREALITY_V2_2_2 that should also be updated.

closed 2025-03-27 by terjeio

kamil1l, 2025-06-20:
@dannyb131 You need to program the board via ISP pins using an ST-Link.

[image]

dannyb131, 2025-06-21:
@kamil1l 

Thanks,

tried programming with an STLink, and built with the default web builder options, everything appears to upload correctly but still cant communicate with the board via USB can re flash marlin and it works fine.

any suggestions?

Thanks

kamil1l, 2025-06-21:
You need to select the Virtual(UART) connection option.
[image]

Rasvapillu, 2025-08-19:
Hi!
I would love your help with this! <3
I got the firmware up n´ running, everything works except i cant figure out
where the pwm-pin is for laser.. =(
Pls guide me to it ;D
Is it 0-5v or 0-24v? 
My laser psu wants 20khz, is that possible with this card?

Thanks for the help! =)

terjeio, 2025-08-20:
PWM output is on _Noz-_ (PA1) according to the schematic. It is an open drain output so you have to add a resistor to whatever voltage (5V, 24V, ...) you want at the output. 20KHz should be possible - set the frequency with `$33`.