grblHAL/STM32F1xx #35: machine suddenly stops while working

MeKeCNC, 2023-05-16:
I am using the latest updated version of grblhal software. my CNC machine suddenly while stops while  working
[image]

terjeio, 2023-05-16:
Noise on the reset input? Is it left floating? Use `$LEV` to check the last state reported.

MeKeCNC, 2023-05-16:
$LEV
[LASTEVENTS:,,,,]
before run.. no events

ALARM:3
GrblHAL 1.1f ['$' or '$HELP' for help]
[MSG:'$H'|'$X' to unlock]
[VER:1.1f.20230507:]
[OPT:VNMZSL,35,512,3,0]
[AXS:3:XYZ]
[NEWOPT:ENUMS,RT+,HOME,TC,SED,TMC=7,SD]
[FIRMWARE:grblHAL]
[NVS STORAGE:*EEPROM]
[DRIVER:STM32F103RC]
[DRIVER VERSION:230429]
[BOARD:BTT SKR MINI E3 V2.0 (alt2)]
[PLUGIN:Trinamic v0.11]
[PLUGIN:SDCARD v1.09]
[PLUGIN:FS macro plugin v0.01]
[G54:0.000,0.000,0.000]
[G55:0.000,0.000,0.000]
[G56:0.000,0.000,0.000]
[G57:0.000,0.000,0.000]
[G58:0.000,0.000,0.000]
[G59:0.000,0.000,0.000]
[G59.1:0.000,0.000,0.000]
[G59.2:0.000,0.000,0.000]
[G59.3:0.000,0.000,0.000]
[G28:0.000,0.000,0.000]
[G30:0.000,0.000,0.000]
[G92:0.000,0.000,0.000]
[HOME:0.000,0.000,0.000:0]
[TLO:0.000,0.000,0.000]
[PRB:0.000,0.000,0.000:0]

after stop
$LEV
[LASTEVENTS:R,,,,]