grblHAL/STM32F1xx #37: how to compiling stm32f103c8t6

hanifhan, 2023-06-05:
[image]

how to compile the stm32f103tx , because the only choice is only for stm32f103rc , if i debug it with rc , the file output is 144 kb which is cant be uploaded to stm32f103c8t6 since the max flash size for f103c8t6 is maximum 128kb, did i need to set the path & symbols ? if yes can u give me a clue the path & symbol please ?

terjeio, 2023-06-05:
TB or CB (128K flash) variants are only debuggable, and barely so, if you [disable USB](https://github.com/grblHAL/STM32F1xx/blob/81f8222fb269e429896e681de419416989783162/Inc/my_machine.h#L36) and use a UART <> bridge to communicate with the controller. Use option _1 Debug_ in the menu for that.

[image]

 I guess I should tag a commit and stop support for 128K variants soon as I will not let these processors hold back further development of the core.

helmihanf, 2023-06-05:
oh i see, actually i want to use stm32f103c8t6 since its cheaper , and i only need the core function , after search it for a while, i found this old driver of yours , and i think its work , still need to test the durability actually , https://github.com/terjeio/grblHAL/tree/master/drivers/STM32F1xx