grblHAL/STM32F1xx #39: Z probe wiring diagram for skr mini e3 v2

brink01, 2023-08-14:
Hello, does anyone have information on how to wire a z probe for the BTT skr mini E3 V2?

terjeio, 2023-08-14:
The probe input is [connected to PC14](https://github.com/grblHAL/STM32F1xx/blob/3f52b06c9d1a8e191ef07673bd4e918e0817177c/Inc/btt_skr_mini_e3_2.0_map.h#L128C38-L128C42) which is E0-STOP [according to the schematic](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/master/hardware/BTT%20SKR%20MINI%20E3%20V2.0/Hardware/BTT%20SKR%20MINI%20E3%20V2.0SCHpdf.PDF).

MeKeCNC, 2023-08-14:
> Hello, does anyone have information on how to wire a z probe for the BTT skr mini E3 V2?

If you want to use Z probe like in marlin, grblHAL probably doesn't support it yet

brink01, 2023-08-15:
> The probe input is [connected to PC14](https://github.com/grblHAL/STM32F1xx/blob/3f52b06c9d1a8e191ef07673bd4e918e0817177c/Inc/btt_skr_mini_e3_2.0_map.h#L128C38-L128C42) which is E0-STOP [according [...]

Thanks. That's what I thought, but there isn't a signal/probe being recognized when it is connected to PC14 and GND. I have tried to invert $6 to see if that makes a difference and it doesn't

terjeio, 2023-08-15:
> I have tried to invert $6 to see if that makes a difference and it doesn't

With nothing connected to the input does $6 change the reported status of the pin? `P` or no `P` in the `Pn:` element in the realtime status report returned by sending `?` (I do not have a board to test with).

FYI [ioSender](https://github.com/terjeio/ioSender) has "LED"s for showing the status of the inputs.

brink01, 2023-08-15:
> With nothing connected to the input does $6 change the reported status of the pin? `P` or no `P` in the `Pn:` element in the realtime status report returned by sending `?` (I do not have a board to [...]

IOsender is a bit new to me so I do apologize. I have turned on "manually validate probe in the settings section. Attempts to validate the probe do not work, and there is no status led on the front page of IOsender to indicate the probe is active. I have also tried to build the Alt- 2 board via web builder to see if that made a difference. So far no luck.

brink01, 2023-08-15:
> > Hello, does anyone have information on how to wire a z probe for the BTT skr mini E3 V2?
> 
> If you want to use Z probe like (BL Touch) in marlin, grblHAL probably doesn't support it yet for F1XX driver

I hope that isn't the case. I have a Scienci Labs auto zero touchplate to use with their gsender software.

terjeio, 2023-08-16:
I've managed to flash a F103C8 MCU with the Mini  board map and PC14 as probe input works for me.
Why it does not for you might be:
* a faulty MCU or board
* incorrect schematics
* a faulty probe or wiring

Is there a voltage on the probe input pin (E0-STOP) when the probe is not connected?
When $6=0 (probe input not inverted) is the P "LED" in ioSender lit? Does it go off when inverted?
Is the probe input working? Test by shorting the pin to ground when the probe is not connected. If this changes the probe "LED" then either your wiring or probe is faulty.

A BL Touch type probe is controlled by PWM to deploy/retract it, this is not what you have. Support for controlling such probes are in the pipeline.