grblHAL/STM32F1xx #40: Please, help check and suggest modifications CNC3040 map file.

RepRapThailand, 2023-08-30:
Hello,

I use Blue Pill but try to re-map pins difference from CNC3040 and need UART1.  This is a map file that I try to modifications and What other files do I need to edit?

[image]

#if N_ABC_MOTORS > 2 || N_GANGED
#error Axis configuration is not supported!
#endif

#if EEPROM_ENABLE
#error EEPROM plugin not supported!
#endif

#if N_AXIS == 5
#define BOARD_NAME "CNC 3040 5-axis"
#elif N_AXIS == 4
#define BOARD_NAME "CNC 3040 4-axis"
#else
#define BOARD_NAME "CNC 3040"
#endif

// Define step pulse output pins.
#define X_STEP_PORT             GPIOA
#define X_STEP_PIN              5
#define Y_STEP_PIN              7
#define Z_STEP_PORT             GPIOB
#define Z_STEP_PIN              1
#define STEP_OUTMODE            GPIO_BITBAND

// Define step direction output pins.
#define X_DIRECTION_PORT        GPIOA
#define X_DIRECTION_PIN         6 
#define Z_DIRECTION_PORT        GPIOB
#define Y_DIRECTION_PIN         0  
#define Z_DIRECTION_PIN         10 
#define DIRECTION_OUTMODE       GPIO_BITBAND

// Define stepper driver enable/disable output pin.
#define STEPPERS_ENABLE_PORT    GPIOB
#define STEPPERS_ENABLE_PIN     13 

// Define homing/hard limit switch input pins.
#define LIMIT_PORT              GPIOA
#define X_LIMIT_PIN             0 
#define Y_LIMIT_PIN             1  
#define Z_LIMIT_PIN             2  
#define LIMIT_INMODE            GPIO_SHIFT10

// Define ganged axis or A axis step pulse and step direction output pins.
#if N_ABC_MOTORS > 0
#define M3_AVAILABLE
#define M3_STEP_PORT            GPIOB
#define M3_STEP_PIN             11  
#define M3_DIRECTION_PORT       GPIOB
#define M3_DIRECTION_PIN        12  
#define M3_LIMIT_PORT           GPIOA
#define M3_LIMIT_PIN            3  
#endif

// Define ganged axis or B axis step pulse and step direction output pins.
#if N_ABC_MOTORS == 2
#define M4_AVAILABLE
#define M4_STEP_PORT            GPIOB
#define M4_STEP_PIN             14 
#define M4_DIRECTION_PORT       GPIOB
#define M4_DIRECTION_PIN        15  
#define M3_LIMIT_PORT           GPIOA
#define M3_LIMIT_PIN            4  
#endif

  // Define spindle enable and spindle direction output pins.
#define SPINDLE_ENABLE_PORT     GPIOA
#define SPINDLE_ENABLE_PIN      8 
#define SPINDLE_DIRECTION_PORT  GPIOA
#define SPINDLE_DIRECTION_PIN   9 

// Define spindle PWM output pin.
#define SPINDLE_PWM_PORT_BASE   GPIOA_BASE
#define SPINDLE_PWM_PIN         10 

// Define flood and mist coolant enable output pins.
#define COOLANT_FLOOD_PORT      GPIOB
#define COOLANT_FLOOD_PIN       3  
#define COOLANT_MIST_PORT       GPIOA
#define COOLANT_MIST_PIN        15 

// Define user-control controls (cycle start, reset, feed hold) input pins.
#define CONTROL_PORT            GPIOB
#define RESET_PIN                    9  
#define FEED_HOLD_PIN           4  
#define CYCLE_START_PIN         5  
#define CONTROL_PORT            GPIOC
#if SAFETY_DOOR_ENABLE
#define SAFETY_DOOR_PIN         15  
#endif
#define CONTROL_INMODE          GPIO_BITBAND

// Define probe switch input pin.
#define PROBE_PORT              GPIOB
#define PROBE_PIN               8  

#if KEYPAD_ENABLE == 1
#error I2C keypad mode is not supported!
#endif

terjeio, 2023-08-30:
If you rename the file to _my_machine_map.h_  then no other files than _my_machine.h_ needs to be changed. Enable the map file [here](https://github.com/grblHAL/STM32F1xx/blob/9db121e046234fc85b53a693a3743da25085cab7/Inc/my_machine.h#L30).
If you plan to submit a PR _my_machine.h_  and _driver.h_ has to be changed and the map file given a new name.

The FEED_HOLD_PIN and M4_LIMIT_PIN (incorrectly defined as M3_LIMIT_PIN  above) cannot have the same pin number since the STM32 architecture only allows one interrupt per pin number regardless of the port. M3_LIMIT_PORT has also a duplicate definition.

FYI the C8 (128K flash version, in reality a CB device) is nearly at its limits for both flash and RAM and updates will soon no longer be possible, consider switching to a medium density (RC or better) variant or F303/F411 Blackpills.

RepRapThailand, 2023-08-31:
Hi, Terjeio

Thanks a lot for your suggest. I don't know how to code but look at other map files and try to fix them. I edited it as you suggested.
Is this correct?
1. FEED_HOLD_PIN  >> PC14
2. M4_LIMIT_PIN >> PA4

terjeio, 2023-08-31:
> Is this correct?

I believe so, compile and flash - then test.

RepRapThailand, 2023-09-04:
I will try today.

RepRapThailand, 2023-09-05:
New error
[image]

terjeio, 2023-09-05:
Your grbl folder is nearly empty, it should contain all the [core files](https://github.com/grblHAL/core/tree/43691fca19c163cb231c655f9e55ff4b4fcb29cb).

RepRapThailand, 2023-09-06:
Newest error
try to fix

[image]

terjeio, 2023-09-06:
Oops, I forgot to check for allowed PWM pins. [Replace these lines](https://github.com/grblHAL/STM32F1xx/blob/5df71fd29a6f4d38d51aa899d220961360df0fad/Inc/driver.h#L117-L141) with the code below to add PA10 PWM output:

```
#if SPINDLE_PWM_PORT_BASE == GPIOA_BASE
  #if SPINDLE_PWM_PIN == 1 // PA1 - TIM5_CH2
    #define SPINDLE_PWM_TIMER_N     5
    #define SPINDLE_PWM_TIMER_CH    2
    #define SPINDLE_PWM_TIMER_INV   0
    #define SPINDLE_PWM_AF_REMAP    0
  #elif SPINDLE_PWM_PIN == 8 // PA8 - TIM1_CH1
    #define SPINDLE_PWM_TIMER_N     1
    #define SPINDLE_PWM_TIMER_CH    1
    #define SPINDLE_PWM_TIMER_INV   0
    #define SPINDLE_PWM_AF_REMAP    0
  #elif SPINDLE_PWM_PIN == 10 // PA10 - TIM1_CH3
    #define SPINDLE_PWM_TIMER_N     1
    #define SPINDLE_PWM_TIMER_CH    3
    #define SPINDLE_PWM_TIMER_INV   0
    #define SPINDLE_PWM_AF_REMAP    0
  #elif SPINDLE_PWM_PIN == 1 // PA1 - TIM2_CH2
    #define SPINDLE_PWM_TIMER_N 2
    #define SPINDLE_PWM_TIMER_CH 2
    #define SPINDLE_PWM_TIMER_INV 0
    #define SPINDLE_PWM_AF_REMAP 0
  #endif
#elif SPINDLE_PWM_PORT_BASE == GPIOB_BASE
  #if SPINDLE_PWM_PIN == 0 // PB0 - TIM1_CH2N
    #define SPINDLE_PWM_TIMER_N     1
    #define SPINDLE_PWM_TIMER_CH    2
    #define SPINDLE_PWM_TIMER_INV   1
    #define SPINDLE_PWM_AF_REMAP    0b01
  #endif
#endif
```

RepRapThailand, 2023-09-07:
> Your grbl folder is nearly empty, it should contain all the [core files](https://github.com/grblHAL/core/tree/43691fca19c163cb231c655f9e55ff4b4fcb29cb).

RepRapThailand, 2023-09-07:
Some error

[image]

[image]

terjeio, 2023-09-07:
`Y_STEP_PORT` and `Y_DIRECTION_PORT` are missing in the map file.
 
Is the board map above the version you are compiling with? I cannot see issues with the IRQ enabled pins in that.

RepRapThailand, 2023-09-07:
> `Y_STEP_PORT` and `Y_DIRECTION_PORT` are missing in the map file.
> 
> Is the board map above the version you are compiling with? I cannot see issues with the IRQ enabled pins in that.

https://github.com/grblHAL/STM32F1xx/archive/refs/heads/master.zip
https://github.com/grblHAL/core/archive/refs/tags/20230714.zip

terjeio, 2023-09-07:
Add your latest copy of the modified map file to a comment.

RepRapThailand, 2023-09-07:
> Add your latest copy of the modified map file to a comment.

/*
  my_machine_map.h - driver code for STM32F103xx ARM processors
  
  Part of grblHAL

  Grbl is free software: you can redistribute it and/or modify
  it under the terms of the GNU General Public License as published by
  the Free Software Foundation, either version 3 of the License, or
  (at your option) any later version.

  Grbl is distributed in the hope that it will be useful,
  but WITHOUT ANY WARRANTY; without even the implied warranty of
  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
  GNU General Public License for more details.

  You should have received a copy of the GNU General Public License
  along with Grbl.  If not, see <http://www.gnu.org/licenses/>.
*/

#if N_ABC_MOTORS > 2 || N_GANGED
#error Axis configuration is not supported!
#endif

#if EEPROM_ENABLE
#error EEPROM plugin not supported!
#endif

#if N_AXIS == 5
#define BOARD_NAME "My Machine 5-axis"
#elif N_AXIS == 4
#define BOARD_NAME "My Machine 4-axis"
#else
#define BOARD_NAME "My Machine"
#endif

// Define step pulse output pins.
#define X_STEP_PORT             GPIOA
#define X_STEP_PIN              5 
#define Y_STEP_PORT             GPIOA
#define Y_STEP_PIN              7 
#define Z_STEP_PORT             GPIOB
#define Z_STEP_PIN              1 
#define STEP_OUTMODE            GPIO_BITBAND

// Define step direction output pins.
#define X_DIRECTION_PORT        GPIOA
#define X_DIRECTION_PIN         6 
#define Y_DIRECTION_PORT        GPIOB
#define Y_DIRECTION_PIN         0  
#define Z_DIRECTION_PORT        GPIOB
#define Z_DIRECTION_PIN         10 
#define DIRECTION_OUTMODE       GPIO_BITBAND

// Define stepper driver enable/disable output pin.
#define STEPPERS_ENABLE_PORT    GPIOB
#define STEPPERS_ENABLE_PIN     13 

// Define homing/hard limit switch input pins.
#define LIMIT_PORT              GPIOA
#define X_LIMIT_PIN             0  
#define Y_LIMIT_PIN             1  
#define Z_LIMIT_PIN             2  
#define LIMIT_INMODE            GPIO_SHIFT10

// Define ganged axis or A axis step pulse and step direction output pins.
#if N_ABC_MOTORS > 0
#define M3_AVAILABLE
#define M3_STEP_PORT            GPIOB
#define M3_STEP_PIN             11  
#define M3_DIRECTION_PORT       GPIOB
#define M3_DIRECTION_PIN        12  
#define M3_LIMIT_PORT           GPIOA
#define M3_LIMIT_PIN            3  
#endif

// Define ganged axis or B axis step pulse and step direction output pins.
#if N_ABC_MOTORS == 2
#define M4_AVAILABLE
#define M4_STEP_PORT            GPIOB
#define M4_STEP_PIN             14  
#define M4_DIRECTION_PORT       GPIOB
#define M4_DIRECTION_PIN        15  
#define M4_LIMIT_PORT           GPIOA
#define M4_LIMIT_PIN            4  
#endif

  // Define spindle enable and spindle direction output pins.
#define SPINDLE_ENABLE_PORT     GPIOA
#define SPINDLE_ENABLE_PIN      8 
#define SPINDLE_DIRECTION_PORT  GPIOA
#define SPINDLE_DIRECTION_PIN   9 

// Define spindle PWM output pin.
#define SPINDLE_PWM_PORT_BASE   GPIOA_BASE
#define SPINDLE_PWM_PIN         10 

// Define flood and mist coolant enable output pins.
#define COOLANT_FLOOD_PORT      GPIOB
#define COOLANT_FLOOD_PIN       3  
#define COOLANT_MIST_PORT       GPIOA
#define COOLANT_MIST_PIN        15 

// Define user-control controls (cycle start, reset, feed hold) input pins.
#define CONTROL_PORT            GPIOB
#define RESET_PIN               2  
#define FEED_HOLD_PIN           4  
#define CYCLE_START_PIN         5  
#define CONTROL_PORT            GPIOC
#if SAFETY_DOOR_ENABLE
#define SAFETY_DOOR_PIN         15  
#endif
#define CONTROL_INMODE          GPIO_BITBAND

// Define probe switch input pin.
#define PROBE_PORT              GPIOB
#define PROBE_PIN               8  

#if KEYPAD_ENABLE == 1
#error I2C keypad mode is not supported!
#endif

RepRapThailand, 2023-09-07:
> There is a duplicate `#define CONTROL_PORT` with conflicting definition. `Z_LIMIT_PIN` and `RESET_PIN` cannot have the same pin number. `M4_LIMIT_PIN` and `FEED_HOLD_PIN` cannot have the same pin number, but it does not matter if motor 4 is not enabled. Ref. the _<interrupt enabled pins> cannot have the same pin number since the STM32 architecture only allows one interrupt per pin number regardless of the port_ comment above. Control and limit pins are interrupt enabled.

#if N_ABC_MOTORS > 2 || N_GANGED
#error Axis configuration is not supported!
#endif

#if EEPROM_ENABLE
#error EEPROM plugin not supported!
#endif

#if N_AXIS == 5
#define BOARD_NAME "My Machine 5-axis"
#elif N_AXIS == 4
#define BOARD_NAME "My Machine 4-axis"
#else
#define BOARD_NAME "My Machine"
#endif

// Define step pulse output pins.
#define X_STEP_PORT             GPIOA
#define X_STEP_PIN              5 // XP /15--74HC245 PIN 18
#define Y_STEP_PORT             GPIOA
#define Y_STEP_PIN              7 // YP /17--74HC245 PIN 16
#define Z_STEP_PORT             GPIOB
#define Z_STEP_PIN              1 // ZP /19--74HC245 PIN 14
#define STEP_OUTMODE            GPIO_BITBAND

// Define step direction output pins.
#define X_DIRECTION_PORT        GPIOA
#define X_DIRECTION_PIN         6  // XD /16--74HC245 PIN 17
#define Y_DIRECTION_PORT        GPIOB
#define Y_DIRECTION_PIN         0  // YD /18--74HC245 PIN 15
#define Z_DIRECTION_PORT        GPIOB
#define Z_DIRECTION_PIN         10 // ZD /21--74HC245 PIN 13
#define DIRECTION_OUTMODE       GPIO_BITBAND

// Define stepper driver enable/disable output pin.
#define STEPPERS_ENABLE_PORT    GPIOB
#define STEPPERS_ENABLE_PIN     13 // /26--U2 74HC245 PIN 18

// Define homing/hard limit switch input pins.
#define LIMIT_PORT              GPIOA
#define X_LIMIT_PIN             0  // IN1 /10--
#define Y_LIMIT_PIN             1  // IN2 /11--
#define Z_LIMIT_PIN             2  // IN3 /12--
#define LIMIT_INMODE            GPIO_SHIFT10

// Define ganged axis or A axis step pulse and step direction output pins.
#if N_ABC_MOTORS > 0
#define M3_AVAILABLE
#define M3_STEP_PORT            GPIOB
#define M3_STEP_PIN             11  // A-XIS /22--74HC245 PIN 12
#define M3_DIRECTION_PORT       GPIOB
#define M3_DIRECTION_PIN        12  // A-XIS /25--74HC245 PIN 111
#define M3_LIMIT_PORT           GPIOA
#define M3_LIMIT_PIN            3  // IN4 /13--
#endif

// Define ganged axis or B axis step pulse and step direction output pins.
#if N_ABC_MOTORS == 2
#define M4_AVAILABLE
#define M4_STEP_PORT            GPIOB
#define M4_STEP_PIN             14  // B-XIS /27--74HC245 PIN 17
#define M4_DIRECTION_PORT       GPIOB
#define M4_DIRECTION_PIN        15  // B-XIS /28--74HC245 PIN 16
#define M4_LIMIT_PORT           GPIOA
#define M4_LIMIT_PIN            4  // IN5 /14--
#endif

  // Define spindle enable and spindle direction output pins.
#define SPINDLE_ENABLE_PORT     GPIOA
#define SPINDLE_ENABLE_PIN      8 // OUT1 /29--74HC245 PIN 15
#define SPINDLE_DIRECTION_PORT  GPIOA
#define SPINDLE_DIRECTION_PIN   9 // OUT2 /30--74HC245 PIN 14

// Define spindle PWM output pin.
#define SPINDLE_PWM_PORT_BASE   GPIOA_BASE
#define SPINDLE_PWM_PIN         10 // PWM /31--

// Define flood and mist coolant enable output pins.
#define COOLANT_FLOOD_PORT      GPIOB
#define COOLANT_FLOOD_PIN       3  // OUT3 /39--74HC245 PIN 11
#define COOLANT_MIST_PORT       GPIOA
#define COOLANT_MIST_PIN        15 // OUT4 /38--74HC245 PIN 12

// Define user-control controls (cycle start, reset, feed hold) input pins.
#define CONTROL_PORT            GPIOB
#define CYCLE_START_PIN         5  // 12pin IDC, x10 /41--
#define CONTROL_PORT            GPIOC
#define RESET_PIN               13  // 12pin IDC, z /2--
#define FEED_HOLD_PIN           14  // 12pin IDC, y /3--
#if SAFETY_DOOR_ENABLE
#define SAFETY_DOOR_PIN         15  // 12pin IDC, x /4--
#endif
#define CONTROL_INMODE          GPIO_BITBAND

// Define probe switch input pin.
#define PROBE_PORT              GPIOB
#define PROBE_PIN               8  // 12pin IDC, x1 /45--

#if KEYPAD_ENABLE == 1
#error I2C keypad mode is not supported!
#endif

RepRapThailand, 2023-09-07:
[image]

[image]

terjeio, 2023-09-07:
Seems like you are building a debug version, this will not or barely fit flash depending on the options enabled in _my_machine.h_.
The release version is smaller and will allow you to enable more options. Select the _Release_ build type from the tool icon (hammer).
Here is how it looks when I built with no options enabled other than your board map:

[image]

PS: I removed the duplicate `#define CONTROL_PORT` definition before compiling.

Tip: to reclaim flash disable USB and use an external USB <> UART breakout board for talking to the controller.

RepRapThailand, 2023-09-11:
I still trying...

terjeio, 2023-09-11:
> I still trying...

Here is a release binary that I built with your map file:

[GRBL Driver STM32F103C8.zip](https://github.com/grblHAL/STM32F1xx/files/12574486/GRBL.Driver.STM32F103C8.zip)