grblHAL/STM32F1xx #41: no usb com port

mikey2023123, 2023-10-10:
Hi i have uploaded the firmware soon as i do that the com port closes, still has power threw the usb but no connection or pc picking it up.
 i have tried to use both firmware's for the SKR MINI E3 V2.0 with no luck any ideas?

terjeio, 2023-10-11:
Local build (Eclipse?) or Web Builder build?
Flashing via SD card?

mikey2023123, 2023-10-11:
I used pio to compile it then flashed it with sd card,I tried all the builds and get the same,I tried using the Web build and I keep getting error 79 codes, will try eclipse today see how that goes will update soon as.

terjeio, 2023-10-11:
If it is not a custom build use the [Web Builder](http://svn.io-engineering.com:8080/)...

BTW did you modify platformio.ini to use the [bootloader loader script](https://github.com/grblHAL/STM32F1xx/blob/master/STM32F103RCTX_BL28K_FLASH.ld)? When flashing via SD card it is required to set addresses in the binary correctly.

closed 2023-12-15 by terjeio