grblHAL/STM32F1xx #42: USB is not working on stm32f103RCT6.

CHTUZKI, 2023-10-25:
HI.

I am using an STM32F103RCT6 development board, I have not made any changes to the project code, I only modified the platformio.ini file, then compiled the project, and used Jink to upload the firmware to my STM32F103RCT6. However, when I plug in the USB cable to my PC, my PC tells me 'Windows has stopped this device because it has reported problems. (Code 43)'. I am not sure why this is happening.

The following is the code for my platformio.ini file:

```
# This is the configuration file for PlatformIO, a cross-platform tool and IDE
# to build and develop software for embedded systems.
#
# To build the firmware with PlatformIO, install PlatformIO Core (CLI, no GUI):
# - https://docs.platformio.org/en/latest//core/installation.html
#
# For documentation on this file, see:
# - https://docs.platformio.org/en/latest/projectconf/index.html
#
# To add support for another STM32F1xx board, add a new [env:name..] section
# with the appropriate `board` and (optionally) `board_build.ldscript`.
# For a list of supported hardware, see:
# - https://docs.platformio.org/en/latest/platforms/ststm32.html
#
# To add support for additional grblHAL plugins or drivers, make the necessary
# changes to the common.build_flags, common.lib_deps and common.lib_extra_dirs
# below.
#
# By default this file sets the preprocessor directive `OVERRIDE_MY_MACHINE`
# to ignore all settings in `Inc/my_machine.h`.  Instead, you should edit this
# file and add the necessary directives (e.g. `-D FOOBAR_ENABLE`) to either
# `common.build_flags` or the environment specific `build_flags`.
#
# Typical command line usage:
#
#    # Build firmware for all defined environments
#    pio run
#    # Build only for specific environments with the -e option
#    pio run -e BTT_SKR_MINI_E3_V20 -e BTT_SKR_MINI_E3_V20_USB
#    # List the freshly built firmware ELF (.elf) and binary (.bin) files:
#    ls -l .pio/build/*/firmware.*
#    # If the target supports uploading (via `upload_protocol`):
#    pio run -e bluepill_f103c8_128k -t upload
#    # Clean up build related resources
#    pio run -t clean
#
[platformio]
include_dir = Inc
src_dir = Src

[common]
build_flags =
  -I .
  -I FatFs
  -I Middlewares/ST/STM32_USB_Device_Library/Class/CDC/Inc
  -I Middlewares/ST/STM32_USB_Device_Library/Core/Inc
  -I USB_DEVICE/Target
  -D RX_BUFFER_SIZE=512
  -D NVS_SIZE=1536
  # Ignore all settings in Inc/my_machine.h (and instead use the below build flags)
  -D OVERRIDE_MY_MACHINE
  # Uncomment to enable debug builds
  #-D DEBUG
  # Uncomment to enable support for keypad
  #-D KEYPAD_ENABLE=1
  # Uncomment to enable support for odometer
  #-D ODOMETER_ENABLE=1
  # Uncomment to enable support for running G-code from the microSD card
  # You also need to uncomment FatFs and sdcard in lib_deps (see below)
  #-D SDCARD_ENABLE=1
lib_deps =
  grbl
  keypad
  motors
  odometer
# To enable support for SD card uncomment `FatFs` and `sdcard` below.
  #FatFs
  #sdcard
  # USB serial support
  Core
  Class
  App
  Target
lib_extra_dirs =
  .
  FatFs
  Middlewares/ST/STM32_USB_Device_Library
  USB_DEVICE

[env]
platform = ststm32
framework = stm32cube
# Do not produce .a files for lib deps (which would prevent them from overriding weak symbols)
lib_archive = no
lib_ldf_mode = off

; [env:BTT_SKR_MINI_E3_V20]
; board = genericSTM32F103RC
; board_build.ldscript = STM32F103RCTX_FLASH.ld
; build_flags = ${common.build_flags}
;   # See Inc/my_machine.h for options
;   -D BTT_SKR_MINI_E3_V20=
;   -D USB_SERIAL_CDC=0
;   # Relocate the vector table where the boot loader expects to find them
;   -D VECT_TAB_OFFSET=0x7000
;   -Wl,--defsym=LD_VECT_TAB_OFFSET=0x7000
; lib_deps = ${common.lib_deps}
;   eeprom
;   trinamic
; lib_extra_dirs = ${common.lib_extra_dirs}
; # Upload is not supported for this board since BOOT0 is tied to GND.
; # With the default boot loader, you must deploy new firmware by copying
; # .pio/build/<env name>/firmware.bin (produced by `pio run`) to the SD card.

[env:BTT_SKR_MINI_E3_V20_USB]
board = genericSTM32F103RC
board_build.ldscript = STM32F103RCTX_FLASH.ld
build_flags = ${common.build_flags}
  # See Inc/my_machine.h for options
  -D BTT_SKR_MINI_E3_V20=
  -D USB_SERIAL_CDC=1
  # Relocate the vector table where the boot loader expects to find them
  -D VECT_TAB_OFFSET=0x7000
  -Wl,--defsym=LD_VECT_TAB_OFFSET=0x7000
lib_deps = ${common.lib_deps}
  eeprom
  trinamic
lib_extra_dirs = ${common.lib_extra_dirs}
upload_protocol = jlink
# Upload is not supported for this board since BOOT0 is tied to GND.
# With the default boot loader, you must deploy new firmware by copying
# .pio/build/<env name>/firmware.bin (produced by `pio run`) to the SD card.

; [env:bluepill_f103c8_128k]
; board = bluepill_f103c8_128k
; board_build.ldscript = STM32F103C8TX_FLASH.ld
; build_flags = ${common.build_flags}
;   # See Inc/my_machine.h for options
;   -D BOARD_CNC3040=
;   -D USB_SERIAL_CDC=0
; lib_deps = ${common.lib_deps}
;   eeprom
; lib_extra_dirs = ${common.lib_extra_dirs}
; upload_protocol = stlink

; [env:bluepill_f103c8_128k_USB]
; board = bluepill_f103c8_128k
; board_build.ldscript = STM32F103C8TX_FLASH.ld
; build_flags = ${common.build_flags}
;   # See Inc/my_machine.h for options
;   -D BOARD_CNC3040=
;   -D NO_SETTINGS_DESCRIPTIONS=
;   -D USB_SERIAL_CDC=1
; lib_deps = ${common.lib_deps}
;   eeprom
; lib_extra_dirs = ${common.lib_extra_dirs}
; upload_protocol = stlink
```

terjeio, 2023-10-25:
1. Does the board have a bootloader you do not want to overwrite? If not you do not want to relocate the vector table. If it has one is the offset correct? And have you modified the loader script to match it?

2. Does the main oscillator crystal frequency match the build? IIRC the default is 8 MHz.

CHTUZKI, 2023-10-25:
I commented out the following code: 
```
-D VECT_TAB_OFFSET=0x7000 
-Wl,--defsym=LD_VECT_TAB_OFFSET=0x7000, 
```
but the USB still doesn't work, I'm not sure what the reason is.

STM32F103RCTX_FLASH.ld and STM32F103RCTX_BL28K_FLASH.ld, what are the roles of these two files? There were no such files in the previous versions. I wanted to use my own .ld file with platformio, so I deleted these two files, and the program encountered an error.

terjeio, 2023-10-25:
> but the USB still doesn't work, I'm not sure what the reason is.

USB is sensitive to clock setup, it has to be exact or it fails. Could be due to the STM32F103xE symbol not beeing defined? See the attached Web Builder .ini file.

>  what are the roles of these two files? 

It tells the linker where to place/find the different elements (code, vector table, static data, ram...) in memory.

If you flash a [Web Builder](http://svn.io-engineering.com:8080/) image does that work? Try with "Mach3 USB breakout (BSMCE04U)". Here is the .ini used for such a build:

[job.zip](https://github.com/grblHAL/STM32F1xx/files/13162359/job.zip)

CHTUZKI, 2023-10-25:
problem solved. job.zip works fine.

closed 2023-10-25 by CHTUZKI