grblHAL/STM32F1xx #43: Motor Stepper buzzes using A4988 and drv 8825

adepacarmerah, 2023-11-12:
I successfully installed GRBLHal on the BOB Mach3 board, but the Stepper Dynamo buzzes and vibrates a bit when using the A4988 driver and drv 8825, but it can run GCode perfectly, is there any solution to prevent the Stepper dynamo from vibrating and buzzing?

terjeio, 2023-11-14:
Is the buzzing there when the motors are not moving and you disable motors off by setting `$1=255`? And doees it change a bit when adjusting the current?
If so it could be normal for this kind of drivers.

adepacarmerah, 2023-11-15:
yes it happens when the motor is idle, the default setting $S1=10 when I replace 255 there is no change, will this hum damage the motor?  If it doesn't damage me, I'll just continue because I really like the GRBLHal, on the Mach 3 board, Oh yes, I intend to build an STM 32 f4xx for 3 axes, do you think I should use a CNC 4060 or at least a Blackpill?

terjeio, 2023-11-16:
> $S1=10 when I replace 255 there is no change

That is odd. When set to 10 the motor drivers should be disabled. However, the driver enable signal is not available on a connector so the drivers are always on with this board...
You write "hum" - is the noise low frequency or high frequency? The drivers you mentioned will, IIRC, sometimes generate a high frequency noise in the motors - more so if the current is set a bit too high.
Anyhow, if the motors does not get too hot I guess it does not matter. If they do be sure to adjust the current to match the motors.

> do you think I should use a CNC 4060 or minimal Blackpill?

Use the one that provides the inputs and outputs you need.

adepacarmerah, 2023-11-17:
thx Terje, I appreciate your work, especially since the newest Sienci longboard will use GRBLHal

closed 2023-11-17 by adepacarmerah