grblHAL/STM32F1xx #44: error:7 - A settings read failed. Auto-restoring affected settings to default values.

jimfong1, 2023-11-13:
Running STM32F103RCT6 on a red rnr MACH3BOB.  Working fine but the only issue is once in awhile I'll get 
error:7 - A settings read failed. Auto-restoring affected settings to default values.
on power up and it loads all the default settings again.. I than have to load them up. Is there a setting to prevent restoring the defaults?   
Thanks Jim

[VER:1.1f.20230903:]
[OPT:VNMSL,35,1024,4,0]
[AXS:4:XYZA]
[NEWOPT:ENUMS,RT+,ES,SED]
[FIRMWARE:grblHAL]
[NVS STORAGE:*FLASH]
[DRIVER:STM32F103RC]
[DRIVER VERSION:230828]
[BOARD:MACH3 USB BOB (BSMCE04U)]
[AUX IO:0,1,0,0]
[CLUSTER:16]
[PLUGIN:LightBurn clusters v0.04]
[G54:0.000,0.000,0.000,0.000]
[G55:0.000,0.000,0.000,0.000]
[G56:0.000,0.000,0.000,0.000]
[G57:0.000,0.000,0.000,0.000]
[G58:0.000,0.000,0.000,0.000]
[G59:0.000,0.000,0.000,0.000]
[G59.1:0.000,0.000,0.000,0.000]
[G59.2:0.000,0.000,0.000,0.000]
[G59.3:0.000,0.000,0.000,0.000]
[G28:0.000,0.000,0.000,0.000]
error:7 - A settings read failed. Auto-restoring affected settings to default values.
[MSG:Warning: Settings write failed!]

terjeio, 2023-11-14:
> Working fine but the only issue is once in awhile I'll get
error:7 - A settings read failed. Auto-restoring affected settings to default values.

This happens at random on bootup? Or while running and on writing to the flash? Note that writing to flash is also done when changing offsets, not only when changing settings.

> Is there a setting to prevent restoring the defaults?

No, restoring defaults happens when a checksum check fails due to data beeing corrupted. So disabling restore is not a good idea. A way around it will be to change the defaults to match your configuration.

See [this discussion](https://github.com/grblHAL/core/discussions/390) for more.

terjeio, 2023-11-15:
> This maybe a clone STM32 processor that was remarked and not authentic? Bad flash???

Well, this board is the only one that I have had reports about settings failures - so a clone issue? The BOB I have at home has a genuine STM32, I fried the clone so I had to replace the MCU. When I am back I will run tests with it to see if I can replicate the issue and if it is possible to add a workaround.

But it is really odd if a read fails on startup if there was no failed write before the previous shutdown. This since reading is just a memcpy from flash to the RAM buffer.

jimfong1, 2023-11-15:
Ok great, Thanks for looking into this.

terjeio, 2023-11-16:
[This comment](https://github.com/grblHAL/core/discussions/390#discussioncomment-7579187) might indicate that it is a clone issue.

terjeio, 2023-12-11:
I have now tested with my MACH3BOB that has a genuine STM MCU and I have yet to see a settings read or write failure. Will continue testing...

jimfong1, 2023-12-12:
Thank for looking into this Terje

Ill probably order a STM32 chip from digikey and replace mine when I get a chance.