grblHAL/STM32F1xx #45: Build fails, not enough flash

terjeio, 2023-11-19:
Since it sounds like ill be doing this myself... Any suggestions of where to post regarding an error Im getting when I do the first build with out change in stmcubeide ".rodata' will not fit in region `FLASH'". I imported the stm32f1xx build, and copied the core, then build....... All fresh and no other builds in the work space.

_Originally posted by @customsolutions in https://github.com/grblHAL/core/discussions/203#discussioncomment-7608942_

terjeio, 2023-11-19:
_Debug_ builds for 128K flash variants is no longer possible. Select the _Release_ build instead.

customsolutions, 2023-11-19:
Thanks that did it, and i was able to use my board file and compile.... but
because im in my office, not out the shop connected to my machine im
getting error: 79 form grbl in my lightburn console.....lol i used $22=0 to
disable auto homing for now.

terjeio, 2023-11-19:
[error: 79 form grbl](https://github.com/grblHAL/core/wiki/First-Run-Grbl-Settings)

closed 2023-12-15 by terjeio