grblHAL/STM32F1xx #53: grblHAL probing with BSMCE04 (STM32F1)

PolyEng, 2024-07-30:


terjeio, 2024-07-30:
When I test here probing works as it should (tested by triggering the input manually).
Please enable `Run substatus` in `General > Status` report options and post a complete log of a probing sequence.

PolyEng, 2024-07-30:
> When I test here probing works as it should (tested by triggering the input manually). Please enable _Run substatus_ in _General > Status report options_ and post a complete log of a probing sequence.

rn i have changed the board , it will take some time for me to swap again the board , i'll keep you updated , thanks in advance

CBR900cc, 2024-09-14:
hello @PolyEng I've some issue with probe pin, in my case if I use z-limit pin with inductive sensor, all working fine, but if I connect inductive sensor to pb4 (probe pin) nothing work, board not read any signal of triggered or not... can you share you pinout connections/photos?

terjeio, 2024-09-15:
@CBR900cc If you short the probe input to ground does it change the probe state?

CBR900cc, 2024-09-28:
@terjeio yes, if shorted probe is triggered

terjeio, 2024-09-29:
> can you explain me how wiring it?

Blue to GND, Brown to 5V and Black to the probe input. Since the sensor is rated 6-36V it could be that it fails to pull down the probe input enough when supplied with only 5V? Can you measure the voltage at the probe input when it is triggered vs. not triggered?

CBR900cc, 2024-10-09:
> > can you explain me how wiring it?
> 
> Blue to GND, Brown to 5V and Black to the probe input. Since the sensor is rated 6-36V it could be that it fails to pull down the probe input enough when supplied with only 5V? Can you measure the voltage at the probe input when it is triggered vs. not triggered?

yes, if is not triggered, output 24v if triggered 0.7v

terjeio, 2024-10-10:
> output 24v if triggered

Ouch - the inputs on the header are not protected by optoucouplers, there are only voltage dividers to step down 5V signals to 3.3V. If you are lucky you have not damaged the MCU. Either add an optocoupler, an additional resitor in series with the input or supply the sensor with a voltage that does not exceed the maximum allowed voltage on the MCU pin. IIRC it is 5V tolerant so 12V or less should be ok since the dividers uses 1K (to ground) and 2K to the input.

CBR900cc, 2024-10-10:
ops... now it is connected with a 1k between signal and gnd and 1k between signal and input header, but now if i do a probe cycle, cycle working correct and grbl see probe is triggered or not with iosender (if I use openbuildscontrol that see probe trigger and crashes), but sometimes it crashes and i need to replug usb cable

terjeio, 2024-10-10:
> but sometimes it crashes and i need to replug usb cable

Perhaps you should increase the 1K resistor between signal and input? 2K is better?

CBR900cc, 2024-10-10:
> 2K

thank you, I try it and update you!