grblHAL/STM32F1xx #54: Firmware web builder stm32 nit working

Jamilgalau, 2024-08-17:
hello terje i build firmware via STM32 bluepill web builder, i have connected EStop or Reset, Cycle Start, and Feed Hold to ground but grbl firmware can't be run on openbuilds, ugs, and io sender, last year i did it and it worked. thx

terjeio, 2024-08-17:
It works for me (CNC3040), default options:

[image]

The port shows up when you connect the board? And can you connect to the port with the sender?

Jamilgalau, 2024-10-12:
Thank you for the latest firmware update, it can work well and run smoothly on UGS, G Sender, and IO Sender, but unfortunately it cannot be used on the OpenBuilds controller, there are error messages alarm 2 and 9

terjeio, 2024-10-13:
Alarm 2 and 9 is related to soft limits and homing respectively. Error 2 and 9 is related to bad comms and alarm beeing active when sending gcode to the controller. So perhaps disable soft limits and ty again?

Jamilgalau, 2024-10-13:
can't connect with openbuilds from the start, I use STM32 Blue Pill, maybe you can try it...but it's normal in almost all gcode senders..I'm used to opnbuilds controller

terjeio, 2024-10-15:
If you power up the controller and connect with a terminal (like puTTY) what is the response when you send a `?` to get a real time report?