grblHAL/STM32F1xx #56: pinmapping is always generic_map.c

kokospalme, 2024-10-19:
I have a BTT SKR MINI E3 V2.0 board, which I want to use with grblHAL. Therefore I uncomment ```#define BOARD_BTT_SKR_MINI_E3_V20``` in ```my_machine.h```. The code compiles as expected and the board seems do boot, but with the default mapping from ```generic_map.h```. So When I check the mapping, I get this:

```
>>> $pins
[PIN:PB6,Feed hold]
[PIN:PB7,Cycle start]
[PIN:PB5,Emergency stop]
[PIN:PA7,Probe]
[PIN:PB12,X limit min]
[PIN:PB13,Y limit min]
[PIN:PB14,Z limit min]
[PIN:PB15,A limit min]
[PIN:PB8,Aux in 0,P0]
[PIN:PA15,Aux in 1,P1]
[PIN:PA0,X step]
[PIN:PA1,Y step]
[PIN:PA2,Z step]
[PIN:PA3,A step]
[PIN:PA4,X dir]
[PIN:PA5,Y dir]
[PIN:PA6,Z dir]
[PIN:PA7,A dir]
[PIN:PA15,Steppers enable]
[PIN:PB1,Spindle on]
[PIN:PB0,Spindle direction]
[PIN:PA8,Spindle PWM]
[PIN:PB3,Mist]
[PIN:PB4,Flood]
[PIN:PA10,RX,Primary UART]
[PIN:PA9,TX,Primary UART]
ok
```

Even when no board is selected, the code compiles, and I get the same pinmapping:
```
// NOTE: Only one board may be enabled!
// If none is enabled pin mappings from generic_map.h will be used.
//#define BOARD_CNC3040
//#define BOARD_CNC_BOOSTERPACK
//#define BOARD_MACH3_BOB
// #define BOARD_BTT_SKR_MINI_E3_V20
//#define BOARD_BTT_SKR_MINI_E3_V20_ALT2
//#define BOARD_SVM
//#define BOARD_CREALITY_V2_2_2 // Not tested!
//#define BOARD_CREALITY_V2_2_7 // Not tested!
// #define BOARD_MY_MACHINE // Add my_machine_map.h before enabling this!
```

I already tried to copy and paste everything from ```btt_skr_mini_e3_2.0_map.h```to ```generic_map.h``, but this caused a whole bunch of errors...

Did I miss something here?
Thank you in advance

terjeio, 2024-10-19:
Can see anything immieadtely wrong, and I do not have access to a board to test with.
Can you try by moving the contents of [line 110](https://github.com/grblHAL/STM32F1xx/blob/2fe1be8f63b6ce28799c3c7f09633c752f0f3dbc/Inc/driver.h#L110) to [line 124](https://github.com/grblHAL/STM32F1xx/blob/2fe1be8f63b6ce28799c3c7f09633c752f0f3dbc/Inc/driver.h#L124)?

> And did I install it correct?

Did you fetch the project recursively so that the submodules was included?

kokospalme, 2024-10-19:
thanks for the quick reply. suprisingly the define ```BOARD_BTT_SKR_MINI_E3_V20``` is not found in ```driver.h```.
When I swap line 110 to line 124 I get the following error:
```
In file included from Inc/i2c.h:25:0,
                 from Src/i2c.c:24:
Inc/driver.h:255:2: error: #error Trinamic plugin not supported!
 #error Trinamic plugin not supported!
  ^~~~~
In file included from Src/diskio.c:15:0:
Inc/driver.h:255:2: error: #error Trinamic plugin not supported!
 #error Trinamic plugin not supported!
  ^~~~~
In file included from Src/driver.c:31:0:
Inc/driver.h:255:2: error: #error Trinamic plugin not supported!
 #error Trinamic plugin not supported!
  ^~~~~
In file included from Src/btt_skr_mini_e3_2.0.c:24:0:
Inc/driver.h:255:2: error: #error Trinamic plugin not supported!
 #error Trinamic plugin not supported!
  ^~~~~
In file included from motors/trinamic.c:25:0:
Inc/driver.h:255:2: error: #error Trinamic plugin not supported!
 #error Trinamic plugin not supported!
  ^~~~~
*** [.pio/build/BTT_SKR_MINI_E3_V20/src/i2c.o] Error 1
*** [.pio/build/BTT_SKR_MINI_E3_V20/src/diskio.o] Error 1
*** [.pio/build/BTT_SKR_MINI_E3_V20/src/btt_skr_mini_e3_2.0.o] Error 1
In file included from USB_DEVICE/App/usbd_cdc_if.c:26:0:
Inc/driver.h:255:2: error: #error Trinamic plugin not supported!
 #error Trinamic plugin not supported!
  ^~~~~
Src/driver.c:66:10: fatal error: eeprom/eeprom.h: No such file or directory
 #include "eeprom/eeprom.h"
          ^~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio/build/BTT_SKR_MINI_E3_V20/src/driver.o] Error 1
*** [.pio/build/BTT_SKR_MINI_E3_V20/lib04c/App/usbd_cdc_if.o] Error 1
*** [.pio/build/BTT_SKR_MINI_E3_V20/lib788/motors/trinamic.o] Error 1
```

What do you mean with recursively? I already read the grblHAL wiki, the readmes and other stuff from the grblHAL plugin repository, but I don't really understand, how these plugins are properly installed.

kokospalme, 2024-10-19:
Okay, I just found out that I can clone the repository with ```git clone --recursive <repository-url>``` to get all the submodules. But I get the same compiling error as mentioned above... 
What I don't understand is why the trinamic plugin shouldn't be supported (see [here](https://github.com/grblHAL/STM32F1xx/blob/master/Inc/driver.h#L254)
When I comment this out, I get another error:
```
trinamic/tmc2660.c: In function 'TMC2660_GetCurrent':
trinamic/tmc2660.c:199:12: warning: 'cs' may be used uninitialized in this function [-Wmaybe-uninitialized]
     return cs2rms_2660(driver, cs);
            ^~~~~~~~~~~~~~~~~~~~~~~
Compiling .pio/build/BTT_SKR_MINI_E3_V20/src/stm32f1xx_hal_msp.o
Compiling .pio/build/BTT_SKR_MINI_E3_V20/src/stm32f1xx_it.o
Compiling .pio/build/BTT_SKR_MINI_E3_V20/src/usb_serial.o
Linking .pio/build/BTT_SKR_MINI_E3_V20/firmware.elf
.pio/build/BTT_SKR_MINI_E3_V20/src/driver.o: In function `driver_init':
driver.c:(.text.driver_init+0x390): undefined reference to `board_init'
collect2: error: ld returned 1 exit status
*** [.pio/build/BTT_SKR_MINI_E3_V20/firmware.elf] Error 1
```

So I don't understand where the problem could be.

kokospalme, 2024-10-25:
> Can see anything immieadtely wrong, and I do not have access to a board to test with. Can you try by moving the contents of [line [...]

But can you compile the code? If yes, would you like to put the project folder in a .zip and put it in conversation? I would really like to compare it to my own to find the error. 
Many thanks

terjeio, 2024-10-25:
Are you compiling with the STM32CubeIDE or PlatformIO?
If the latter then the platformio.ini file is not correct, `-D BTT_SKR_MINI_E3_V20=` should be `-D BOARD_BTT_SKR_MINI_E3_V20=`
Somebody with PlatformIO installed should check/update platformio.ini and submit a PR for it.

terjeio, 2024-10-25:
FYI I just [added this to the Wiki](https://github.com/grblHAL/core/wiki/Compiling-grblHAL#platformio).

kokospalme, 2024-10-25:
Many thanks! [That](https://github.com/grblHAL/STM32F1xx/issues/56#issuecomment-2437219728) helped.

closed 2024-10-25 by kokospalme

kokospalme, 2024-11-20:
> Are you compiling with the STM32CubeIDE or PlatformIO?

sorry I overlooked your question. I'm compiling with PlatformIO.

terjeio, 2024-11-21:
> I'm compiling with PlatformIO.

Then [my_machine.h is not used](https://github.com/grblHAL/STM32F1xx/blob/44755b3ff8915002efe562291637e37005fa4598/platformio.ini#L1C1-L36C2), options has to be specified in the platformio.ini file. Or remove/comment out `-D OVERRIDE_MY_MACHINE` to use _my_machine.h_ anyway