grblHAL/STM32F1xx #57: Support for STM32 clones?

jake5253, 2024-11-15:
I apologize in advance if this has been covered previously -- its even possible it has been deemed as not worth the additional maintenance, which would be understandable.
Also, as a disclaimer, I'll admit right up front that I don't know what I'm doing when it comes to C.
I have 2 spare 3D printers (and controllers) laying around my workshop. The controllers are both Voxelab-branded, but they're essentially the same as Creality 4.4.2 boards -- except they were built during the international chip shortage crisis so one of them has a Nations Tech 'N32G455' MCU and the other has a HUADA 'HC32F460' MCU.  I attempted to research how to shoehorn the N32 into working order, but ultimately it was a fail. 
Between STM32 and N32, there was differences in the names/references of timers, I think the N32 was lacking the CAN peripheral, interrupts were handled differently, and there was something about the system clock 'HSE' I think? Unfortunately, I don't recall -- or understand entirely what the implications of that all is.
I didn't bother doing anything on the HUADA side, since what information I found made it seem as though that would require more changes than N32.

If anyone could assist me in getting grblHAL running on either of those boards, I would be thrilled! I'm willing to sacrifice one or both boards to testing, as they're both just collecting dust anyways.
Even if someone could point me in the direction of some good resources of learning how to get this working on my own, that would be helpful and appreciated also.

Below is the "SDK" repos for both MCUs. 
[hc32f46x_ddl](https://github.com/wellrun/hc32f460/tree/master/hc32f460_sdk/%E9%A9%B1%E5%8A%A8%E5%BA%93%E5%8F%8A%E6%A0%B7%E4%BE%8B/hc32f46x_ddl)
[N32G455xx](https://github.com/NationsTechCoreLib/N32G455xx/)
Possibly also of interest, I just found HC32 for PlatformIO [here](https://github.com/shadow578/platform-hc32f46x)

Thank you very much for your consideration!

terjeio, 2024-11-16:
I suggest starting with a copy [STM32F3xx code](https://github.com/grblHAL/STM32F3xx) since this is the most basic one.

Having access to reference manuals, at least for the MCU, and preferably for the SDK depending on how it is structured - is a must.
Good general knowledge of how a MCU work is a plus, there is a lot of material available on the web for those who want to learn.

jake5253, 2024-11-17:
Thank you, @terjeio I will look at the STM32F3 codebase.  That hadn't actually occurred to me since these chips are "drop-in replacements" for 'F1 chips.  The reference manuals seem to be fairly available, and using my browser's built-in Translate feature, CSDN appears to be a valuable resource of more technical documents than one could ever dream of.  I have, at best, a vague understanding of how MCUs work, registers, stacks, pointers.. but not in-depth. So, I'll see what I can find online; surely there's several good "How MCUs work" guides I'll have to locate to get a better understanding, and probably somewhere in that process I'll have to decide if its worth the effort for these boards.