grblHAL/STM32F1xx #58: grblHAL couldn't communicate with stepper driver on BTT SKR mini e3 v2.0 (and other problems)

kokospalme, 2024-11-18:
I have a fresh cloned repository (with all submodules of course), but I ran into some issues. 
The goal:
- a 4-axis CNC with the btt skr mini e3 v2.0 board
- swap Y and Z axis, as the boards default "double" axis is Z, but my machine (the famous "dremel cnc" project) has two y axes

the problems:
- the board doesn't seem to communicate with the stepper driver (see log)
- the motors kind of "click" when I want to move them, but they don't turn in any direction

What I did in the code so far:
- my_machine.h: uncomment line 27: ```#define BOARD_BTT_SKR_MINI_E3_V20```
- config.h: line 42: ```#define N_AXIS 4 // Number of axes```

This is my console log (incl. settings):
```
>>> $$
$0 = 10.0    (Step pulse time, microseconds)
$1 = 25    (Step idle delay, milliseconds)
$2 = 0    (Step pulse invert, mask)
$3 = 0    (Step direction invert, mask)
$4 = 15    (Invert step enable pin, boolean)
$5 = 0    (Invert limit pins, boolean)
$6 = 0    (Invert probe pin, boolean)
$9 = 1   
$10 = 511    (Status report options, mask)
$11 = 0.010    (Junction deviation, millimeters)
$12 = 0.002    (Arc tolerance, millimeters)
$13 = 0    (Report in inches, boolean)
$14 = 0   
$15 = 0   
$16 = 0   
$17 = 0   
$18 = 0   
$19 = 0   
$20 = 0    (Soft limits enable, boolean)
$21 = 0    (Hard limits enable, boolean)
$22 = 0    (Homing cycle enable, boolean)
$23 = 0    (Homing direction invert, mask)
$24 = 25.0    (Homing locate feed rate, mm/min)
$25 = 500.0    (Homing search seek rate, mm/min)
$26 = 250    (Homing switch debounce delay, milliseconds)
$27 = 1.000    (Homing switch pull-off distance, millimeters)
$28 = 0.100   
$29 = 0.0   
$30 = 1000.000    (Maximum spindle speed, RPM)
$31 = 0.000    (Minimum spindle speed, RPM)
$32 = 0    (Laser-mode enable, boolean)
$33 = 5000.0   
$34 = 0.0   
$35 = 0.0   
$36 = 100.0   
$37 = 0   
$39 = 1   
$40 = 0   
$43 = 1   
$44 = 4   
$45 = 3   
$46 = 0   
$47 = 0   
$62 = 0   
$63 = 3   
$64 = 0   
$65 = 0   
$100 = 250.00000    (X-axis travel resolution, step/mm)
$101 = 250.00000    (Y-axis travel resolution, step/mm)
$102 = 250.00000    (Z-axis travel resolution, step/mm)
$103 = 250.00000   
$110 = 500.000    (X-axis maximum rate, mm/min)
$111 = 500.000    (Y-axis maximum rate, mm/min)
$112 = 500.000    (Z-axis maximum rate, mm/min)
$113 = 500.000   
$120 = 10.000    (X-axis acceleration, mm/sec^2)
$121 = 10.000    (Y-axis acceleration, mm/sec^2)
$122 = 10.000    (Z-axis acceleration, mm/sec^2)
$123 = 10.000   
$130 = 200.000    (X-axis maximum travel, millimeters)
$131 = 200.000    (Y-axis maximum travel, millimeters)
$132 = 200.000    (Z-axis maximum travel, millimeters)
$133 = 200.000   
$140 = 500   
$141 = 500   
$142 = 500   
$143 = 500   
$150 = 16   
$151 = 16   
$152 = 16   
$153 = 16   
$180 = 25.0   
$181 = 25.0   
$182 = 25.0   
$183 = 25.0   
$190 = 500.0   
$191 = 500.0   
$192 = 500.0   
$193 = 500.0   
$200 = 22.0   
$201 = 22.0   
$202 = 22.0   
$203 = 22.0   
$210 = 50   
$211 = 50   
$212 = 50   
$213 = 50   
$220 = 22.0   
$221 = 22.0   
$222 = 22.0   
$223 = 22.0   
$339 = 0   
$341 = 0   
$342 = 30.0   
$343 = 25.0   
$344 = 200.0   
$345 = 200.0   
$346 = 1   
$370 = 0   
$376 = 0   
$384 = 0   
$394 = 4.0   
$398 = 100   
$481 = 0   
$484 = 1   
$486 = 0   
$538 = 0   
$673 = 1.0   
ok
>>> $J=G21G91X100F100
ok
GrblHAL 1.1f ['$' or '$HELP' for help]
[MSG:Warning: Could not communicate with stepper driver!]
>>> $J=G21G91Y100F100
ok
GrblHAL 1.1f ['$' or '$HELP' for help]
[MSG:Warning: Could not communicate with stepper driver!]
>>> $J=G21G91Z-2.007F100
ok
GrblHAL 1.1f ['$' or '$HELP' for help]
[MSG:Warning: Could not communicate with stepper driver!]
>>> $J=G21G91A1F100
ok
GrblHAL 1.1f ['$' or '$HELP' for help]
[MSG:Warning: Could not communicate with stepper driver!]

```

What could be the problem here?

I already got the project to work,

kokospalme, 2024-11-19:
update: I tried [this](https://github.com/grblHAL/STM32F1xx/issues/49#issuecomment-2057697265), just to see what happens, and now the steppers show a very odd behavior, similar to the mentioned post. I also swapped the Y and Z axes pins in the code.
When I move a single axis, all steppers kind of "tick" for a millisecond. The X-axis moves, but vibrates a lot when doing so. Same for the y-axis. The z-axis just vibrates and does not move, maybe because there is not enough torque. The A-axis is not yet build, but the single motor does move, but less then it should. 

Does anyone have a clue where to start troubleshooting?

terjeio, 2024-11-19:
> in btt_skr_mini_e3_2.0_map.h SERIAL1_PORT is defined with 31. Why is this 31?

31 is [UART3 with alt. 1 pin mappings](https://github.com/grblHAL/STM32F1xx/blob/357904dcce3947c0f3e8f8d886770396ef33e0e7/Inc/serial.h#L33). Which is used to communicate with the drivers according to the schematic.

> I also swapped the Y and Z axes pins in the code.

This is not enough, you will have to swap the driver address in the [Trinamic datagrams](https://github.com/grblHAL/STM32F1xx/blob/master/Src/btt_skr_mini_e3_2.0.c) as well if Y and Z has different settings.

Did you try [this patch](https://github.com/grblHAL/STM32F1xx/issues/49#issuecomment-2057697265)?

kokospalme, 2024-11-19:
> 31 is [UART3 with alt. 1 pin mappings](https://github.com/grblHAL/STM32F1xx/blob/357904dcce3947c0f3e8f8d886770396ef33e0e7/Inc/serial.h#L33). Which is used to communicate with the drivers according to the schematic.
thank you, that makes sense :)

> Did you try [this patch](https://github.com/grblHAL/STM32F1xx/issues/49#issuecomment-2057697265)?
Yes, I already tried that. With this M122 reports the status of the motors, but the odd behavior didn't stop.

> This is not enough, you will have to swap the driver address in the [Trinamic datagrams](https://github.com/grblHAL/STM32F1xx/blob/master/Src/btt_skr_mini_e3_2.0.c) as well if Y and Z has different settings.
Allright, I'll try this tomorrow and see where I can change the address.

terjeio, 2024-11-20:
> Yes, I already tried that. With this M122 reports the status of the motors, but the odd behavior didn't stop.

Are you sure the motor wiring is correct?

Here is the status output I get, with a STM32F446 board, and the motor is running fine:

```
                      X
Driver          TMC2209
Set current         500
RMS current         490
Peak current        692
Run current       15/31
Hold current       7/31
CS actual          7/31
PWM scale             9
vsense          1=0.180
stealthChop        true
msteps               16
tstep           1048575
pwm
threshold             0
[mm/s]                0
OT prewarn        false
OT prewarn has
been triggered    false
off time              3
blank time            1
hysteresis
-end                  3
-start                1
Stallguard thrs      22
DRIVER STATUS:
stallguard     
sg_result             6
fsactive               
stst                  *
olb                    
ola                    
s2gb                   
s2ga                   
otpw                   
ot                     
STATUS REGISTERS:
 X = 0xC0:07:00:00
```

terjeio, 2024-11-20:
Tip: [here is how I change the address](https://github.com/grblHAL/STM32F4xx/blob/ffa25639089726f3c60937a397c3eeac7e51a91b/Src/tmc_uart.c#L458-L471) when drivers have designated UART ports. You can reference the axis number in the motors parameter to switch the address around.

terjeio, 2024-11-20:
I changed the address by modifying the config parameter (config->address, passed by reference), the motor parameter is passed by value and cannot be changed. axis = id for the primary axis motor, axis != id for the secondary. The address range for TMC2209 is 0 - 3 and has to be set in the datagrams when the drivers share a single UART. The example I provided sets the address to 0 for all drivers since each driver has a separate UART channel (the two address pins on the driver chips then are set to 0 for all).

kokospalme, 2024-11-20:
That's interesting, because I seem to get the right motor even without [this](https://github.com/grblHAL/STM32F1xx/issues/58#issuecomment-2488735708) or your modification of code for STM32F4 boards. For example when I unplug the Y-axis motor (former z axis), M122 reveals that the Y-axis motor is actually unplugged (variables ola, olb):

```
>>> m122
[TRINAMIC]
                      X       Y       Z
Driver          TMC2209 TMC2209 TMC2209
Set current         500     500     500
RMS current         489     489     489
Peak current        691     691     691
Run current       15/31   15/31   15/31
Hold current       7/31    7/31    7/31
CS actual          7/31    7/31    7/31
PWM scale             9       9       9
vsense          1=0.180 1=0.180 1=0.180
stealthChop       false   false   false
msteps               16      16      16
tstep           1048575 1048575 1048575
pwm
threshold             0       0       0
[mm/s]                0       0       0
OT prewarn        false   false   false
OT prewarn has
been triggered    false   false   false
off time              3       3       3
blank time            0       0       0
hysteresis
-end                 -1      -1      -1
-start                1       1       1
Stallguard thrs       0       0       0
DRIVER STATUS:
stallguard     
sg_result             2       2       2
fsactive                               
stst                  *       *       *
olb                   *       *       *
ola                           *        
s2gb                                   
s2ga                                   
otpw                                   
ot                                     
STATUS REGISTERS:
 X = 0x80:07:00:80
 Y = 0x80:07:00:C0
 Z = 0x80:07:00:80
ok
```

I tested this with all 3 motors, and they were all referenced right. Or is there something I didn't understand yet? I already read the TMC2209 datasheets and studied the grblHAL code and the schematic of the btt board.

terjeio, 2024-11-21:
> ... and the schematic of the btt board.

If you look closely Y and Z are addressed as 2 and 1 respectively. So perhaps the schematic is correct and Y and Z should be swapped as default? I do not have a board so I cannot verify this.

FYI the M122 ola/olb status can only be detected reliably if M122 is sent when the motors are running.

bmajkut, 2025-06-01:
Hello,
I installed grblHAL from web builder on SKR mini e3 v1.2 and I am getting the same error.
@kokospalme did You managed to fix that issue?

terjeio, 2025-12-14:
Check/compare [M122 output](https://github.com/grblHAL/Plugins_motor/tree/a7552435ce08edeebe009e55cd230fab07a4a692#m122---output-debug-info-or-reset-driver) after $RST and reboot.