grblHAL/STM32F1xx #59: can't compile for stm32 F1XX web builder

adepacarmerah, 2024-11-25:
Hi Terje, for the past 2 days I have not been able to compile grblhal stm32f1xx for cnc 3040 via web builder, it says build failed

terjeio, 2024-11-28:
Hmm, the compiler used by the Web Builder does not optimize as well as the one used by the STM32CubeIDE so the flash overflows. Either I have to remove Web Builder support for the cnc3040 or I'll have to remove functionality, possibly the canned cycle gcodes?

Jamilgalau, 2024-11-28:
remove some rarely used functions, the important thing is to be solid to do GRBL work, because this board is easy to get and provides a low cost controller option

terjeio, 2024-11-29:
The Web Builder has been updated, I stripped out a bunch of pin names so it should compile for a while longer.

closed 2024-11-29 by terjeio