grblHAL/STM32F1xx #63: grblhal runs normally without any problems on stm32f1xx 64kb

FusionHolic, 2024-12-24:
hello terje, i flashed cnc 3040 firmware to stm32 bluepill 64kb because the seller sent the wrong item. but it was successfully flashed and i ran the cnc machine without any problems for more than a week running an average of 5 hours per day, is this normal? will there be no problems in the future

terjeio, 2024-12-25:
Then the MCU is a 128K device - I would be surprised if grblHAL would run at all if only half of the code is present in flash.

FusionHolic, 2024-12-25:
everything works smoothly including the homing system and limit switch, the cnc finishes the job without any problems, you might need to try it, btw I just bought a stm32 fx1xx with a type C connection, the seller said 128 KB but the flash size turned out to be 256kb, are there many clone stm32s circulating that do not match the specifications?

terjeio, 2024-12-25:
The STM1xx series surely is cloned in various ways, more or less compatible with the original. It is always best to ensure you get a real ST chip, not a clone.
AFAIK if it is a 256K device it is a high-density version (xC, xD or xE), what does the chip marking say?
BTW more grblHAL features can be added when running on a high-density device, all boards available in the Web Builder except CNC 3040 are for high-density.

closed 2025-01-16 by FusionHolic