grblHAL/STM32F1xx #64: Defining b motor for 4 axis board

Tony-g80, 2025-01-01:
Hi everyone,

A bit off course as it’s for an “unsupported” so to speak board. 

I have repurposed my old creality V2.4.s1 board from an ender 7 ( variant of 4.2.2  STMF103Ret6 ) the board itself uses external dm542 for x and y and TMC 2208 in standalone mode for Z and E, I have function and control of all motors so that parts great and working. 

My issue is with the designation of the fourth stepper, I can only get it defined as an A motor, ideally I would like to look at a B motor but down the road I want it mounted on the tool head so I can get some non planar motions on the x axis when in use, sounds weird probably but years ago when I watched a video of someone making their own mould plugs for fibreglassing they did it with a Cartesian style machine and the 4th axis was on the tool head on the B plane with non planar milling motions on the x axis. 

so far I can only get my 4th axis to show as an A axis no matter what, my number of available axis is defined in config.h as 4, if I leave the M3 available stemmed and pins/ports in place I have a working rotaryA axis but if I comment out the whole M3 available statement and associated pins/ports it won’t allow me to replace with M4 available in its place for that group of settings it just has a fit that M3 isn’t available, I have put a #define B_axis in my machine pin map and also tried with it in the config.h

Am I grasping at straws here and will only ever be able to have an A axis on my fourth stepper with a 4 driver board or is their other settings that I’ve not managed to find, I have spent the last week or so trying to find enough info to get past this hurdle but it’s a bit impossible at times to find the info I’m looking for, can anyone shed some light on if I can get my wayward machine config working. 

Thanks.

Tony-g80, 2025-01-01:
Just as reference this is where I want to assign my b motor but I can’t get it to configure then4th available axis, not my design but the route I would like to go. 

[image]

terjeio, 2025-01-02:
Currently it is not possible to assign axis letters to motors in an easy way, it is something I am considering.
It is possible if you modify the code though, changing `A` to `B` [here](https://github.com/grblHAL/core/blob/5a052010418d5f49ffa42114713ff7de56362ff4/gcode.c#L1401) and [here](https://github.com/grblHAL/core/blob/5a052010418d5f49ffa42114713ff7de56362ff4/nuts_bolts.c#L68) _may_ work.

Tony-g80, 2025-01-02:
Hi terjeio, 

Thanks for the response, that seems nice and straightforward thankfully.

I will hopefully get a chance later today to have another play and recompile/flash this and see how it goes, I will report back. 

Many thanks and happy new year to you. 😁