grblHAL/STM32F1xx #65: BTT SKR Mini E3 V2.0 problem with limit switches

kokospalme, 2025-01-02:
I'm trying to get the limit switches working, but I don't get a signal on the pins.
When I short the pins from the limit switch connector, the signal doesn't change. I always get 0V at the node where for example R1 and C1 are connected. When I run the setup wizard from UGS and select "invert limit switches", X is turning red, and turns green when I unselect it. Y and Z are always green, no matter if I connect the pins from the connector of if I invert the signal. 
in ```boards/btt_skr_mini_e3_2.0_map.h``` the pins are:
```
// Define homing/hard limit switch input pins.
#define LIMIT_PORT              GPIOC
#define X_LIMIT_PIN             0 //PC0
#define Y_LIMIT_PIN             1 //PC1
#define Z_LIMIT_PIN             2 //PC2
#define LIMIT_INMODE            GPIO_BITBAND
```
[image]

Is my board defective?

terjeio, 2025-01-03:
What are the voltages on J1-J3 pin 1 if you press and hold the reset switch (or short pin 5 to ground in the SWD connector - J11) before powering up the board and while measuring?

kokospalme, 2025-01-03:
I always get 0V, no matter if RST is short to gnd or not. Only a voltage of about 700mV appears for a couple of seconds when I switch off the whole power supply.

terjeio, 2025-01-03:
Then I am afraid the MCU is damaged, from the PRM:

_During and just after reset, the alternate functions are not active and the I/O ports are configured in Input Floating mode ..._

This on the assumption that the pullup resistors (R1, R3, R5, R8) are ok and 3.3V is present on their "top" side.

kokospalme, 2025-01-03:
Thanks for the support! I measured the supply voltages. 3.3V should be okay. I just ordered a new board, but also a new CPU and see what I can do with my reflow gun and flashing via STlink refering to [this thread](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/issues/431). It would be sad to throw away the whole board just because of a faulty cpu.