grblHAL/STM32F1xx #67: 4 Axis for Stm32 F1xx 256kb

FusionHolic, 2025-01-16:
I have already designed many boards for stm32 F1XX, it works well with 3 axes, I want to install 4 axes but there is no facility in the web builder, I use stm32 F1XX 256 kb
[image]

terjeio, 2025-01-16:
Which chip exactly since it is 256K? You want to use the CNC 3040 map?

FusionHolic, 2025-01-16:
yes i want to use f1xx blue pills with map 3040 for 4 Axis, Karen let me design with that map

terjeio, 2025-01-17:
So the MCU chip is marked "f1xx"?

FusionHolic, 2025-01-17:
Yes...f1xx, i send pic stllink utility

[image]

FusionHolic, 2025-01-17:
thank you very much terje, great work and great dedication

closed 2025-01-17 by FusionHolic