grblHAL/STM32F1xx #68: add support for stm32l152rct6 256kb 32kb ram +true  8kb eeprom

0voltex0, 2025-03-30:
hi i want to know how easy o hard to add support for its cortex m3 stm32l152rct6 running at 48mhz slightly overclocked over its max 32mhz
with it internal 8kb eeprom and it 256 kb 32kb ram  it's good enough as hardware for grbl i have a bench of them and  i want to make pcb for a laser engraver with it 
help and guid me please

0voltex0, 2025-03-30:
yes it has one 32bit timer  (TIM5) i am using eide extension on vscode i did delete (driver and  Middleware, Startup ) and replaced them with stm32lxx file , i did chose stm32f1xx as template not stmf3xx
 i will see what can i do on my own thanks for replay

0voltex0, 2025-03-31:
hi friend i get the project to compile after some trial and error know when i connect to bc win10 i get (USB CDC connections device descriptor request failed) maybe is related to my system clock 

void SetSysClock48MHZ(void)
{
  __IO uint32_t StartUpCounter = 0, HSEStatus = 0;
 
  /* SYSCLK, HCLK, PCLK2 and PCLK1 configuration ---------------------------*/
  /* Enable HSE */
  RCC->CR |= ((uint32_t)RCC_CR_HSEON);
 
  /* Wait till HSE is ready and if Time out is reached exit */
  do
  {
    HSEStatus = RCC->CR & RCC_CR_HSERDY;
    StartUpCounter++;
  } while((HSEStatus == 0) && (StartUpCounter != HSE_STARTUP_TIMEOUT));
 
  if ((RCC->CR & RCC_CR_HSERDY) != RESET)
  {
    HSEStatus = (uint32_t)0x01;
  }
  else
  {
    HSEStatus = (uint32_t)0x00;
  }
 
  if (HSEStatus == (uint32_t)0x01)
  {
    /* Enable 64-bit access */
    FLASH->ACR |= FLASH_ACR_ACC64;
 
    /* Enable Prefetch Buffer */
    FLASH->ACR |= FLASH_ACR_PRFTEN;
 
    /* Flash 1 wait state */
    FLASH->ACR |= FLASH_ACR_LATENCY;
 
    /* Enable the PWR APB1 Clock */
    RCC->APB1ENR |= RCC_APB1ENR_PWREN;
 
    /* Select the Voltage Range 1 (1.8V) */
    PWR->CR = PWR_CR_VOS_0;
 
    /* Wait Until the Voltage Regulator is ready */
    while((PWR->CSR & PWR_CSR_VOSF) != RESET)
    {
    }
 
    /* HCLK = SYSCLK */
    RCC->CFGR |= (uint32_t)RCC_CFGR_HPRE_DIV1;
 
    /* PCLK2 = HCLK */
    RCC->CFGR |= (uint32_t)RCC_CFGR_PPRE2_DIV1;
 
    /* PCLK1 = HCLK */
    RCC->CFGR |= (uint32_t)RCC_CFGR_PPRE1_DIV1;
 
    /*  PLL configuration: PLLCLK = (HSE * 12) / 2 = 48MHz //(HSE * 16) / 3 = 64MHz  */
    RCC->CFGR &= (uint32_t)((uint32_t)~(RCC_CFGR_PLLSRC | RCC_CFGR_PLLMUL |RCC_CFGR_PLLDIV));                               
    RCC->CFGR |= (uint32_t)(RCC_CFGR_PLLSRC_HSE | RCC_CFGR_PLLMUL12 | RCC_CFGR_PLLDIV2);
 
    /* Enable PLL */
    RCC->CR |= RCC_CR_PLLON;
 
    /* Wait till PLL is ready */
    while((RCC->CR & RCC_CR_PLLRDY) == 0)
    {
    }
 
    /* Select PLL as system clock source */
    RCC->CFGR &= (uint32_t)((uint32_t)~(RCC_CFGR_SW));
    RCC->CFGR |= (uint32_t)RCC_CFGR_SW_PLL;
 
    /* Wait till PLL is used as system clock source */
    while ((RCC->CFGR & (uint32_t)RCC_CFGR_SWS) != (uint32_t)0x0C)
    {
    }
            /* Configure USB Clock */
        #if defined(RCC_CFGR_USBPRE)
            RCC->CFGR &= ~RCC_CFGR_USBPRE; // USB (48 MHz)
        #endif
           // Configure the USB clock
          /* Enable USB Clock */
        #if defined(RCC_APB1ENR_USBEN)
            RCC->APB1ENR |= RCC_APB1ENR_USBEN; // Enable USB clock
        #elif defined(RCC_APB1ENR1_USBEN)
            RCC->APB1ENR1 |= RCC_APB1ENR1_USBEN; // Enable USB clock (for some STM32 families)
        #endif

        /* Optional: Enable USB Interrupts */
        NVIC_EnableIRQ(USB_LP_IRQn); // Enable USB low-priority interrupt
  }
  else
  {
    /* If HSE fails to start-up, the application will have wrong clock */
            /* If HSE fails to start, fall back to HSI (16 MHz) */
        /* Enable HSI */
        RCC->CR |= RCC_CR_HSION;
        while ((RCC->CR & RCC_CR_HSIRDY) == 0)
        {
        }

        /* Configure Flash latency for 16 MHz */
        FLASH->ACR &= ~FLASH_ACR_LATENCY;
        FLASH->ACR |= FLASH_ACR_LATENCY_0;

        /* Select HSI as system clock source */
        RCC->CFGR &= ~RCC_CFGR_SW;
        RCC->CFGR |= RCC_CFGR_SW_HSI;
        while ((RCC->CFGR & RCC_CFGR_SWS) != RCC_CFGR_SWS_HSI)
        {
        }
  }
}

0voltex0, 2025-04-02:
i'm still not able to pinpoint the problem (my knowledge is limited)  using jlink as debugger  i found that the code stack at this function -( ( memcpy(&hal.stream, stream, sizeof(io_stream_t));  ) 
in local varibels  ( ret = 0 '<error reading variable> ) what its mean?
can you suggest some process to do my diagnostic   if flash emulate is not working correctly can it make usb cdc not detected by win10 correctly , how to determin that grbl configured it self correctely , help me and  thanks .

0voltex0, 2025-04-05:
I spent so much time trying to debug the problem using J-Link, but I had no results. While searching on Google, I came across a guy who had the same problem. He used CubeMX to generate files for USB CDC and was facing the same issue. Using CubeIDE solved his problem, so I generated the necessary files with the startup and updated my project in VSCode, and now it works like a charm! The MCU connects to GSender, sends commands, and everything works. I still have some work to do, but for now, I'm good.

0voltex0, 2025-04-06:
ok
[VER:1.1f.20250329:]
client $$
[OPT:NMSL,100,1024,3,0]
[NEWOPT:ENUMS,RT+,EXPR,SED]
[FIRMWARE:grblHAL]
[SIGNALS:RHS]
[NVS STORAGE:*EEPROM]
[DRIVER:STM32L152RCT]
[DRIVER VERSION:250328]
ok
$0=5.0 (Step pulse time, μs)
$1=25 (Step idle delay, ms)
$2=0 (Step pulse invert, mask)
$3=0 (Step direction invert, mask)
$4=7 (Invert step enable pin, boolean)
$5=0 (Invert limit pins, boolean)
$6=0 (Invert probe pin, boolean)
$9=0
$10=511 (Status report options, mask)
$11=0.010 (Junction deviation, mm)
$12=0.002 (Arc tolerance, mm)
$13=0 (Report in inches, boolean)
$14=0
$15=0
$16=0
$17=0
$18=0
$19=0
$20=0 (Soft limits enable, boolean)
$21=0 (Hard limits enable, mask)
$22=0 (Homing cycle enable, mask)
$23=0 (Homing direction invert, mask)
$24=25.0 (Homing locate feed rate, mm/min)
$25=500.0 (Homing search seek rate, mm/min)
$26=250 (Homing switch debounce delay, ms)
$27=1.000 (Homing switch pull-off distance, mm)
$28=0.100 (G73 pull-off distance, mm)
$29=0.0 (Step Pulse Delay, μs)
$30=1000.000 (Maximum spindle speed, rpm)
$31=0.000 (Spindle Mode, )
$32=0 (Laser-mode enabled as spindle, boolean)
$33=5000.0 (Spindle Frequency, Hz)
$34=0.0 (Spindle Duty Cycle, %)
$35=0.0 (Minimum spindle speed, %)
$36=100.0 (Maximum spindle speed, %)
$37=0 (Deenergized Steppers, %)
$39=1 (Printable Command Characters, boolean)
$40=0 (Soft Limits Jogging, boolean)
$43=1 (Homing Locate Cycle, )
$44=4 (Axis Mask, )
$45=3 (Axis Mask, )
$46=0 (Axis Mask, )
$62=0 (Sleep Function, boolean)
$63=3 (Disable Laser on Hold, boolean)
$64=0 (Alarm on Startup, boolean)
$65=0 (Allow Feedrate Override, boolean)
$100=250.00000 (X-axis travel resolution, step/mm)
$101=250.00000 (Y-axis travel resolution, step/mm)
$102=250.00000 (Z-axis travel resolution, step/mm)
$110=500.000 (X-axis maximum rate, mm/min)
$111=500.000 (Y-axis maximum rate, mm/min)
$112=500.000 (Z-axis maximum rate, mm/min)
$120=10.000 (X-axis acceleration, mm/sec^2)
$121=10.000 (Y-axis acceleration, mm/sec^2)
$122=10.000 (Z-axis acceleration, mm/sec^2)
$130=200.000 (X-axis maximum travel, mm)
$131=200.000 (Y-axis maximum travel, mm)
$132=200.000 (Z-axis maximum travel, mm)
$341=0 (Manual tool change mode, )
$342=30.0 (Probing distance, mm)
$343=25.0 (Probing slow feed rate, mm/min)
$344=200.0 (Probing seek feed rate, mm/min)
$345=200.0
$346=1
$384=0
$394=0.0
$398=100
$481=0
$486=0
$534=1
$539=0.0
$673=0.0
$676=3
ok
ok
ok
ok
ok
0 - PWM
ok
client $$
$0=5.0 (Step pulse time, μs)
$1=25 (Step idle delay, ms)
$2=0 (Step pulse invert, mask)
$3=0 (Step direction invert, mask)
$4=7 (Invert step enable pin, boolean)
$5=0 (Invert limit pins, boolean)
$6=0 (Invert probe pin, boolean)
$9=0
$10=511 (Status report options, mask)
$11=0.010 (Junction deviation, mm)
$12=0.002 (Arc tolerance, mm)
$13=0 (Report in inches, boolean)
$14=0
$15=0
$16=0
$17=0
$18=0
$19=0
$20=0 (Soft limits enable, boolean)
$21=0 (Hard limits enable, mask)
$22=0 (Homing cycle enable, mask)
$23=0 (Homing direction invert, mask)
$24=25.0 (Homing locate feed rate, mm/min)
$25=500.0 (Homing search seek rate, mm/min)
$26=250 (Homing switch debounce delay, ms)
$27=1.000 (Homing switch pull-off distance, mm)
$28=0.100 (G73 pull-off distance, mm)
$29=0.0 (Step Pulse Delay, μs)
$30=1000.000 (Maximum spindle speed, rpm)
$31=0.000 (Spindle Mode, )
$32=0 (Laser-mode enabled as spindle, boolean)
$33=5000.0 (Spindle Frequency, Hz)
$34=0.0 (Spindle Duty Cycle, %)
$35=0.0 (Minimum spindle speed, %)
$36=100.0 (Maximum spindle speed, %)
$37=0 (Deenergized Steppers, %)
$39=1 (Printable Command Characters, boolean)
$40=0 (Soft Limits Jogging, boolean)
$43=1 (Homing Locate Cycle, )
$44=4 (Axis Mask, )
$45=3 (Axis Mask, )
$46=0 (Axis Mask, )
$62=0 (Sleep Function, boolean)
$63=3 (Disable Laser on Hold, boolean)
$64=0 (Alarm on Startup, boolean)
$65=0 (Allow Feedrate Override, boolean)
$100=250.00000 (X-axis travel resolution, step/mm)
$101=250.00000 (Y-axis travel resolution, step/mm)
$102=250.00000 (Z-axis travel resolution, step/mm)
$110=500.000 (X-axis maximum rate, mm/min)
$111=500.000 (Y-axis maximum rate, mm/min)
$112=500.000 (Z-axis maximum rate, mm/min)
$120=10.000 (X-axis acceleration, mm/sec^2)
$121=10.000 (Y-axis acceleration, mm/sec^2)
$122=10.000 (Z-axis acceleration, mm/sec^2)
$130=200.000 (X-axis maximum travel, mm)
$131=200.000 (Y-axis maximum travel, mm)
$132=200.000 (Z-axis maximum travel, mm)
$341=0 (Manual tool change mode, )
$342=30.0 (Probing distance, mm)
$343=25.0 (Probing slow feed rate, mm/min)
$344=200.0 (Probing seek feed rate, mm/min)
$345=200.0
$346=1
$384=0
$394=0.0
$398=100
$481=0
$486=0
$534=1
$539=0.0
$673=0.0
$676=3
ok
client $9=1
ok
client $
[HLP:$$ $# $G $I $N $x=val $Nx=line $J=line $SLP $C $X $H $B ~ ! ? ctrl-x]
ok
client ?
hi my friend terjeio. way i cant get anwer for ? command

terjeio, 2025-04-07:
You have modified the [STM USB code](https://github.com/grblHAL/STM32F1xx/blob/4be4a2a4439af4d1efc47f2911b6fb59b12d3691/USB_DEVICE/App/usbd_cdc_if.c#L230-L235) to check for linestate changes?

0voltex0, 2025-04-08:
"Sorry for not responding earlier, my friend. My HDD died, and I didn’t have a backup, so I had to start from scratch. Anyway, I didn’t change the line state—but now it’s working!

Also, if I add an LCD like the ST7735 and use bit-banged SPI, will there be any bottleneck for my MCU? Thank you, my friend!"

terjeio, 2025-04-09:
If you want to display anything during motion you should use DMA enabled SPI to update the screen.

0voltex0, 2025-04-16:
"Hi my friend, after I finished the firmware, I made a prototype and tested the board. Overall it's working, but I have some missed steps. I checked the timer on STM32F3xx and I see that you're using SystemCoreClock = 8000000U (8 MHz) instead of 72 MHz. and I'm using 48 MHz. I don't know where to start or look for this problem. Can you help please?

terjeio, 2025-04-16:
If you use TIM5 (32 bit) for the stepper time you should use the F3xx code for step generation. And you have to adjust the [timer latency defaults](https://github.com/grblHAL/STM32F3xx/blob/5fb84f7a0c37120ebb49a0113c343317d94be9a0/Inc/driver.h#L96-L113) to match the lower MCU clock, you really need an oscilloscope or logic analyzer to do that properly.
FYI the `$0` setting (step pulse width) has been reduced to 5 us, you may have to increase that back to 10us?

> "I feel like there is some conflict, and some steps for X to go to Y (or vice versa).

Likely the step timing is off, it is highly unlikely that axes get mixed up.

>  can you look at my driver?

For that I would need a board and your code so I can check it with a scope. I cannot tell how the processor will behave by looking at the code only. BTW have you tried to compile with optimization turned on?

terjeio, 2025-04-17:
> I use the default -Og optimization

Try -O3. Moving some critical parts of the code to RAM _may_ help, but perhaps unlikely if there is no penalty (latency) when accessing flash.

Which settings are you testing with? $0, $29, steps/mm and acceleration.