grblHAL/STM32F1xx #74: How to disappear "Reset signal"

yoshi515rei, 2025-07-26:
My controller is BSMCE04U-PP with the latest version grblhal. 
Firmweare is build withStm32CubeIDE. Just changed "N_Axis 3 to 4" and "ENABLE_BACKLASH_COMPENSATION off to on" in the Config.h.
Please find the control signals photo. "R" is no need for my CNC.
I would like to know how to di

[image]

terjeio, 2025-07-27:
Add `#define CONTROL_ENABLE (CONTROL_FEED_HOLD|CONTROL_CYCLE_START)` to [my_machine.h](https://github.com/grblHAL/STM32F1xx/blob/master/Inc/my_machine.h).

yoshi515rei, 2025-07-31:
I'm using iosender iosender2.0.44.
I have installed the latest iosender2.046, and the firmweare which was build without "#define CONTROL_ENABLE (CONTROL_FEED_HOLD|CONTROL_CYCLE_START)" in the my_machine.h has been uploaded.
Results.
"R" is disapped. And there is no issues so far.
Thanks.